bacnet-stack/bacnet-stack #490: epics filter just object list

cyberumut, 2023-09-10:
how can i do epics just filter object list. 
Too many queries are being sent, I just want it to pull a list of objects.

example : bacrp 7 8 7 76  1
I want to get this output on epics

skarg, 2023-09-10:
The Device object object-list property is a BACnetARRAY, which means that array-index 0 is the size of the array (number of elements) and each element is array-index 1..N.  

In some products (i.e. export table), I created a state machine to discover the devices, read the Device object object-list property array-index 0, and then 1..N, and store the values in structure stored in a Keylist for each device, and use that structure when acquiring the values. Those products used bacnet_read_property_queue() to query the property value, and waited on for bacnet_read_write_idle() before continuing or timing out. 

The EPICS example application does something similar.

cyberumut, 2023-09-10:
When I run this code in parallel and read data from two different devices, it stops sending and receiving data after a while. What is the reason?
```
float returnfloat =0.0;
bacnet_data_analog_present_value(readdeviceid, 1,readanalogid, &returnfloat );
```

closed 2023-09-11 by cyberumut

BACnetEd, 2023-09-12:
By "in parallel" do you mean from different threads? I wonder if the
client-side library is thread-safe. I pretty sure that for server-side you
have to put in your own resource access protections, but I do see Steve has
been addressing the issues here and there. Just my 2c.