bacnet-stack/bacnet-stack #508: How can we define static address for objects

coskunyildirimcosel, 2023-10-09:
Hello,
I want to remove the max address limit on objects and statically assign the address I want. I want objects to be added according to the assigned address. How can we do this?

coskunyildirimcosel, 2023-10-09:
Hello Steve,
Thank you for your quick response.
For example, we have a MAX_ANALOG_INPUTS limit on Analog_Input.
Instead I want to assign an object with a random static address.
When I add Analog_Input_Present_Value(55) I want an object with address 55 to be created automatically. Currently when I set Max_analog_inputs to 55 I create objects from 0 to 54 that I don't need.
I hope I could explain.

skarg, 2023-10-09:
There are lots of methods to create static objects at arbitrary instances.  Keep the 'instance' and 'index' consistent with each other so that the device object can iterate through the object list.

There are examples using the 'keylist.c' library to dynamically create objects - see an example [ao.c module](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/object/ao.c),

Here is an example where you can modify the object data structure to include the object_instance:
```
struct object_data {
    const uint32_t Object_Instance;
    const char *Object_Name;
    uint16_t Units;
    float Min_Value;
    float Max_Value;
    bool Writable : 1;
    bool Out_Of_Service : 1;
    float Present_Value;
};
```
Create a static object property value table:
```
/* default values */
static struct object_data Object_List[] = {
    { 55, "Sensor 4A", UNITS_DEGREES_CELSIUS, 0.0, 100.0, true, false, 18.0 }
};
/* number of objects */
static size_t Objects_List_Size = sizeof(Object_List) / sizeof(Object_List[0]);

/**
 * @brief Determines the number of objects N
 * @return  Number of objects N
 */
unsigned Analog_Input_Count(void)
{
    return Objects_List_Size;
}

/**
 * @brief Return the object or NULL if not found.
 * @param  object_instance - object-instance number of the object
 * @return  object if the instance is found, and NULL if not
 */
static struct object_data *Object_List_Element(uint32_t object_instance)
{
    unsigned index;

    for (index = 0; index < Objects_List_Size; index++) {
        if (Object_List[index].object_instance == object_instance) {
            return &Object_List[index];
        }
    }

    return NULL;
}

/**
 * @brief Determines the object instance-number for a given 0..N index
 * of Analog Value objects where N is Analog_Value_Count().
 * @param  index - 0..N value
 * @return  object instance-number for the given index
 */
uint32_t Analog_Input_Index_To_Instance(unsigned index)
{
    uint32_t object_instance = UINT32_MAX;
    
    if (index < Objects_List_Size) {
        object_instance = Object_List[index].object_instance;
    }

    return object_instance;
}

/**
 * @brief For a given object instance-number, determines a 0..N index
 * @param  object_instance - object-instance number of the object
 * @return  index for the given instance-number, or N if not valid.
 */
unsigned Analog_Input_Instance_To_Index(uint32_t object_instance)
{
    unsigned index;

    for (index = 0; index < Objects_List_Size; index++) {
        if (Object_List[index].object_instance == object_instance) {
            break;
        }
    }

    return index;
}

/**
 * @brief For a given object instance-number, gets the present-value
 * @param  object_instance - object-instance number of the object
 * @return  present-value of the object
 */
float Analog_Input_Present_Value(uint32_t object_instance)
{
    float value = 0;
    struct object_data *pObject;

    pObject = Object_List_Element(object_instance);
    if (pObject) {
        value = pObject->value;
    }

    return value;
}
```

coskunyildirimcosel, 2023-10-09:
Ok thank you very much