bacnet-stack/bacnet-stack #509: How to subscribe to multiple cov objects in the same device?

Twillpower, 2023-10-10:
How to subscribe to multiple cov objects in the same device?
```
void wy_updata_io_led_cov(void)
{
     IO_LED_INFO io_led_info = {0};
     struct mstimer COV_Timer;
     mstimer_set(&COV_Timer, 1 * 1000);
     while(1)
     {
     if (mstimer_expired(&COV_Timer)) {
         mstimer_reset(&COV_Timer);
         handler_cov_timer_seconds(mstimer_interval(&COV_Timer)/1000);
     }
         IO_LED_Info_Set(0, io_led_info);

         handler_cov_task();
         vTaskDelay(10/portTICK_PERIOD_MS);
     }
}
```
Create multiple such tasks separately, or
```
void wy_updata_cov(void)
{
     IO_LED_INFO io_led_info = {0};
     AIR_INFO m_air_info_data = {0};
     struct mstimer COV_Timer;
     mstimer_set(&COV_Timer, 1 * 1000);
     while(1)
     {
     if (mstimer_expired(&COV_Timer)) {
         mstimer_reset(&COV_Timer);
         handler_cov_timer_seconds(mstimer_interval(&COV_Timer)/1000);
     }
         IO_LED_Info_Set(0, io_led_info);

         AIR_Info_Set(0, m_air_info_data);
         handler_cov_task();

         vTaskDelay(10/portTICK_PERIOD_MS);
     }
}
```
I am currently testing using a timer to generate fake data, and found that after subscribing to two objects at the same time, the cov notification received is confusing, and the time (lifetime) does not match the timer.

skarg, 2023-10-12:
From the code posted, it appears that the bacnet_task() is running in a separate task from COV without using mutex for data protection with other data in the stack.

In a product I developed that used MSTP with FreeRTOS:

- called all the BACnet tasks functions in the one FreeRTOS task with a thread-safe queue for incoming PDUs and outgoing PDUs
- one task for mstimer
- one task for each MSTP datalink calling datalink_receive() feeding the incoming queue with PDUs received, and sending the outgoing queued PDUs

There is a work-in-progress (incomplete, not tested yet, not compiled yet) branch of a [FreeRTOS BACnet MSTP](https://github.com/bacnet-stack/bacnet-stack/tree/feature/freertos-mstp) example at github.

You might also have success refactoring `handler_cov_fsm()` to suit your needs (it only processes one subscription and one state per call), shrinking the number of COV subscriptions, or running your handler_cov_fsm() task faster than 10ms.

coskunyildirimcosel, 2023-10-17:
Hello,
In version 0.9, Analog_Input and Binary_Input also work with COV and there are no problems.
I noticed that COV is not working in the latest version.
handler_cov_task also causes a stack overflow.
I'm working with Esp32.
If there is a setting, can you share it?

Twillpower, 2023-10-27:
Hello,
My current solution is to use ucov instead of ccov.

coskunyildirimcosel, 2023-10-30:
> Merhaba, Şu anki çözümüm ckov yerine ucov kullanmak.

hello, thank you for your answer.
In which version of COV Bacnet does it work properly?
COV does not work for me in version 1.3.1. In version 0.9, COV responds fast enough in mstp, but in IP it responds in 1-2 seconds.

Twillpower, 2023-11-02:
The version I am currently using is 1.0.0. I only use mstp. IP is not currently used. Both ccov and ucov can run through.