bacnet-stack/bacnet-stack #513: How to reinitialize a bacnet stack on windows

paulgr2008, 2023-10-17:
Hi!
I'm trying to implement some BACnet client features (who-is and readprops) in the GUI application in Windows 11. To do that I should have an opportunity to free resources when I don't need them and reinitialize them if I need them without exiting the main. I try to accomplish that using the datalink_cleanup function after I turn off the Bacnet functionality. When I just start the application and initialize the BACnet stack and the datalink the first time it works perfectly, but for some reason, I don't fully understand, every time I try to REinitialize the datalink I get the same error: "failed to allocate a socket [error code 10093] Winsock not initialized. This message is returned by any function except WSAStartup(), indicating that a successful WSAStartup() has not yet been performed".
What did I miss? How should I proceed? Should I do an additional cleaning?
Thank you in advance!

MightyPork, 2023-10-17:
I am using `datalink_init` and `dataling_cleanup` on Linux and re-init seems to work fine, suggesting this might be something platform specific. I never tested it on Windows.

paulgr2008, 2023-10-17:
> I am using `datalink_init` and `dataling_cleanup` on Linux and re-init seems to work fine, suggesting this might be something platform specific. I never tested it on Windows.
@MightyPork Sorry, but it doesn't help me, because my priority right now is Windows porting. But if the Windows porting exists, somebody has tested it already, I suppose. Or am I wrong? :)

paulgr2008, 2023-10-17:
I've decided to use your own example https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/whois/main.c to reproduce the issue. 
This code works on Linux, but it doesn't work on Windows... :(
```
int main(int argc, char *argv[])
{
    BACNET_ADDRESS src = { 0 }; /* address where message came from */
    uint16_t pdu_len = 0;
    unsigned timeout_milliseconds = 0;
    unsigned delay_milliseconds = 100;
    struct mstimer apdu_timer = { 0 };
    struct mstimer datalink_timer = { 0 };
    long dnet = -1;
    BACNET_MAC_ADDRESS mac = { 0 };
    BACNET_MAC_ADDRESS adr = { 0 };
    BACNET_ADDRESS dest = { 0 };
    bool global_broadcast = true;
    int argi = 0;
    unsigned int target_args = 0;
    char *filename = NULL;
    bool repeat_forever = false;
    long retry_count = 0;

    /* check for local environment settings */
    if (getenv("BACNET_DEBUG")) {
        BACnet_Debug_Enabled = true;
    }
    /* decode any command line parameters */
    filename = filename_remove_path(argv[0]);
    for (argi = 1; argi < argc; argi++) {
        if (strcmp(argv[argi], "--help") == 0) {
            print_usage(filename);
            print_help(filename);
            return 0;
        }
        if (strcmp(argv[argi], "--version") == 0) {
            printf("%s %s\n", filename, BACNET_VERSION_TEXT);
            printf("Copyright (C) 2014 by Steve Karg and others.\n"
                   "This is free software; see the source for copying "
                   "conditions.\n"
                   "There is NO warranty; not even for MERCHANTABILITY or\n"
                   "FITNESS FOR A PARTICULAR PURPOSE.\n");
            return 0;
        }
        if (strcmp(argv[argi], "--mac") == 0) {
            if (++argi < argc) {
                if (bacnet_address_mac_from_ascii(&mac, argv[argi])) {
                    global_broadcast = false;
                }
            }
        } else if (strcmp(argv[argi], "--dnet") == 0) {
            if (++argi < argc) {
                dnet = strtol(argv[argi], NULL, 0);
                if ((dnet >= 0) && (dnet <= BACNET_BROADCAST_NETWORK)) {
                    global_broadcast = false;
                }
            }
        } else if (strcmp(argv[argi], "--dadr") == 0) {
            if (++argi < argc) {
                if (bacnet_address_mac_from_ascii(&adr, argv[argi])) {
                    global_broadcast = false;
                }
            }
        } else if (strcmp(argv[argi], "--repeat") == 0) {
            repeat_forever = true;
        } else if (strcmp(argv[argi], "--retry") == 0) {
            if (++argi < argc) {
                retry_count = strtol(argv[argi], NULL, 0);
                if (retry_count < 0) {
                    retry_count = 0;
                }
[... 105 lines cut ...]
            }
            if (mstimer_expired(&apdu_timer)) {
                if (repeat_forever || retry_count) {
                    Send_WhoIs_To_Network(&dest, Target_Object_Instance_Min,
                        Target_Object_Instance_Max);
                    retry_count--;
                } else {
                    break;
                }
                mstimer_reset(&apdu_timer);
            }
        }
      
        print_address_cache();
 // => my code
        fflush(stdout);
        datalink_cleanup();
        Sleep(1000);
        iii++;
    }
    // =>

    return 0;
}
```

paulgr2008, 2023-10-17:
Actually, I've found the source of the issue! 
```
static void bip_init_windows(void)
{
    static bool initialized = false;
    int Result;
    WSADATA wd;

    if (!initialized) {
        Result = WSAStartup((1 << 8) | 1, &wd);
        /*Result = WSAStartup(MAKEWORD(2,2), &wd); */
        if (Result != 0) {
            print_last_error("TCP/IP stack initialization failed");
            exit(1);
        }
        initialized = true;
        atexit(bip_cleanup);
    }
}
```
Surprisingly, the variable initialized remains true! (but only in Windows). It's because it's static, isn't it? Though it's strange. And I can't find the place where we can reinitilize it back. Could you help me with it?

PR #514 (2023-10-18): Issues/issue 513 how to reinitialize a bacnet stack on windows
Fixes #513 BACnet/IP on Windows by checking the initialized flag to file scope and checking it prior to cleanup.  Similar checking added to BACnet/IPv6 on Windows.
closed 2023-10-18 by skarg