bacnet-stack/bacnet-stack #515: Bacnet testing with BTS

ErikasMasaitis, 2023-10-18:
Hello, i tried to run BTS (https://www.bac-test.com/bacnet-test-suite-bts/), got a lot of failed tests. Wanted to know if there is any plans to test Bacnet stack with BTS or similar test benches?

I am using example bacnet router application:

```
Fail      BC 135.1: 10.2.2.2.1      - Respond to Who-Is-Router-To-Network with I-Am-Router-To-Network when no Network Number specified
Fail      BC 135.1: 10.2.2.2.2-A    - Respond to Who-Is-Router-To-Network with I-Am-Router-To-Network with known Network Number
Fail      BC 135.1: 10.2.2.2.2-B    - Respond to Who-Is-Router-To-Network with I-Am-Router-To-Network with known Network Number - Requires BTT Transponder
Fail      BC 135.1: 10.2.2.2.3-A    - Nearside Network Number is Specified and the Router Does Not Respond
Fail      BC 135.1: 10.2.2.7.1-B    - Reject - Unknown Network
Fail      BC 135.1: 10.2.2.7.2      - Unknown Network Layer Message Type
Fail      BC 135.1: 10.2.3.6.1-B    - Failed Attempt To Locate Downstream Routers
Fail      BC 135.1: 10.2.4.8        - Remote Broadcast to local Network_Number should be ignored
Fail      BC 135.1: 10.2.X2         - Routers Execute What-Is-Network-Number
Fail      BC 135.1: 10.8.2.1.2      - Respond to Who-Is-Router-To-Network with I-Am-Router-To-Network with known Network Number
Fail      BC 135.1: 10.8.2.2.2      - Unknown Network Layer Message Type
Fail      BC 135.1: 10.8.3.5.1      - Unknown Network - Ensure Reject not issued by Router if Broadcast
Fail      BITS    : BIT00300        - Issuing Reject-Message-To-Network Unrecognised Service
Fail      BITS    : BIT00305-B      - Attempt to overwrite Accessible Route in Routing Table with I-Am-Router
Fail      BITS    : BIT00309        - Rediscover all Devices via Remote Broadcast Who-Is
Fail      BITS    : BIT00310        - Router Table vs Accessible Networks test
Fail      BITS    : BIT01300        - Confirm Broadcast Distribution Table has Own IP
Fail      BITS    : BIT01802        - BVLL should ignore some Broadcast messages

Pass      BC 135.1: 10.2.2.7.3-C    - Unknown Network Layer Message Type Directed to FarSide for someone else - No BTT
Pass      BC 135.1: 10.2.2.7.3-D    - Unknown Network Layer Message Type Local Broadcast - No BTT
Pass      BC 135.1: 10.2.2.7.3-E    - Unknown Network Layer Message Type Global Broadcast - No BTT
Pass      BC 135.1: 10.2.2.7.3-F    - Unknown Network Layer Message Type Remote Broadcast - No BTT
Pass      BC 135.1: 14.3.X1         - v17 Write-BDT service is required to return Write-BDT-NAK
Pass      BC 135.1: 14.6.1          - Execute Read Foreign Device Table
Pass      BC 135.1: 14.6.5          - Delete Foreign Device Table Entry Which Should Be Rejected
Pass      BC 135.1: 14.6.6          - Delete Foreign Device Table Entry
Pass      BITS    : BIT00018        - Check Port ID
Pass      BITS    : BIT00302-A      - Check for Duplicate Router Table Entries
Pass      BITS    : BIT00302-B      - Check for Duplicate I-Am-Router Entries
Pass      BITS    : BIT00303        - Check that there is only one (directly connected) Routing Entry per port
Pass      BITS    : BIT00305-A      - Attempt to add Duplicate Routing Entry using Init-Routing-Table
Pass      BITS    : BIT00306        - Directed I-Am with Broadcast DADR should not be reflected
Pass      BITS    : BIT00308        - Rediscover all Routers via Remote Broadcast Who-Is-Router

and a lot of ignored tests.....
```

BACnetEd, 2023-10-18:
Gents

I am concerned about anyone wasting time here - mainly due to "false
positives".

Our BACnet Test Suite has been used to test many (instances of B-ASC/B-BC)
BACnet implementations, including many based on this stack, for
successful BTL certifications.

However, although it has been used for many non-BTL implementations, it has
never been used to test a *router* through BTL... at least not
completely... not yet. However, it is being used at the moment for a
project that is about 95% of the way through BTL router testing, so it is
nearly there.

So, I will check Erikas' list against a 'good' router list and make sure
that the BTS is consistent and provide feedback.

And if anyone has a particular question, or wishes to challenge, any of the
tests, please let me know.

Steve, if you wish us to take this offline, let me know and we can do so.

ErikasMasaitis, 2023-11-06:
Hello again, have you looked at tests?

BACnetEd, 2023-11-06:
Hello Erikas - sorry, I thought I did answer..

I _did_ run all the tests against some commercial routers and the tests all appear to be valid. In other words, no false alarms are suspected. 

Please proceed accordingly. If you do come across an issue that is not clear / seems incorrect, please don't hesitate to contact me on this forum, for the benefit of all, else at edward@bac-test.com.

josephtoutoo, 2024-01-20:
Is anyone did BTL Certification using this stack. if you have information. Could you please provide information.

BACnetEd, 2024-01-21:
Hi Joseph

All I can say here is we have helped multiple customers who have used
this stack get their products through BTL successfully.

BUT - this stack has missing properties here and there, and these need to
be added before submittal. This is straightforward, but it can take a
surprisingly long time to get up the BACnet learning curve (if you are not
there already) and implement them all.

The stack itself is a bit of a moving target, because new objects,
properties and services are continually being added, and the new ones,
because they are new, may have even more missing features. You really have
to draw a line under what Protocol Revision, profile and BIBBs you intend
your product to have, figure out what is missing, implement those, and
go from there. You cannot go too far back in Protocol Revisions (BTL has a
limit: https://bacnetwiki.com/index.php?title=Minimum_Protocol_Revision),
but by no means do you have to implement the very latest.

In terms of 'code quality' of what is in this implementation - no
complaints. Things could be optimised here and there (e.g. event driven
COVs rather than an 'idle loop'), but for what it is - primarily a
simple-as-possible implementation/starting point - it is just fine.

And brace yourself - you are likely in for a $10K++ bill from BTL, even for
a relatively simple device.

Ed
***@***.***
www.bac-test.com

josephtoutoo, 2024-01-21:
Hello Edward,

Could you please mention the customer if you have any info.

Could you please mention the missing properties .

We are looking for protocol revision 19.

We are working on new product.

Do you have any code quality reports - static analysis report.

Cyclo metric complexity.
And other metrics.

Thank you,
Joseph toutoo.

BACnetEd, 2024-01-21:
Joseph
Answers inline
Ed

> Hello Edward,
>
> Could you please mention the customer if you have any info.

EKH: Sorry, we have NDAs in place and can't mention "who"

>
>
> Could you please mention the missing properties .

EKH: I don't have a complete list on-hand. And the properties are always a
combination of what your selection of profile/BIBBS/Protocol Revision.
Our BACnet Test Suite will do this for you. Let me know if you are
interested in purchasing, but offline - let's keep this conversation in the
forum on a non-commercial basis from now on.
Contact me: ***@***.***

>
>
> We are looking for protocol revision 19.

EKH: This is good for now. It will be acceptable by BTL until Jan 2027 per
https://bacnetwiki.com/index.php?title=Minimum_Protocol_Revision

>
>
> We are working on new product.
>
> Do you have any code quality reports - static analysis report.

>
> Cyclo metric complexity.
> And other metrics.

EKH: I don't, but someone else with the project loaded may have. Let's
wait. Bear in mind that a lot of the folks involved in this stack are going
to be at AHR for this coming week.

>
>
> Thank you,
> Joseph toutoo.
>
>
> > Hi Joseph
> >
> > All I can say here is we have helped multiple customers who have used
> > this stack get their products through BTL successfully.
> >
> > BUT - this stack has missing properties here and there, and these need
> to
> > be added before submittal. This is straightforward, but it can take a
> > surprisingly long time to get up the BACnet learning curve (if you are
> not
> > there already) and implement them all.
> >
> > The stack itself is a bit of a moving target, because new objects,
> > properties and services are continually being added, and the new ones,
> > because they are new, may have even more missing features. You really
> have
> > to draw a line under what Protocol Revision, profile and BIBBs you
> intend
> > your product to have, figure out what is missing, implement those, and
> > go from there. You cannot go too far back in Protocol Revisions (BTL has
> a
> > limit: https://bacnetwiki.com/index.php?title=Minimum_Protocol_Revision),
>
> > but by no means do you have to implement the very latest.
> >
> > In terms of 'code quality' of what is in this implementation - no
> > complaints. Things could be optimised here and there (e.g. event driven
> > COVs rather than an 'idle loop'), but for what it is - primarily a
> > simple-as-possible implementation/starting point - it is just fine.
> >
> > And brace yourself - you are likely in for a $10K++ bill from BTL, even
> > for
> > a relatively simple device.
> >
> > Ed
> > www.bac-test.com
> >
> >
> >
> >
> > > Is anyone did BTL Certification using this stack. if you have
> > information.
> > > Could you please provide information.
> > >
> > > <
> >
> https://github.com/bacnet-stack/bacnet-stack/issues/515#issuecomment-1901643555>,
>
> >
> > > <
> >
>
> >
> > >
> >
> >
> >
> >
> >
> > *Edward Hague**BACnet Interoperability Testing Services, Inc*
> > +1 (650) 305 1395
> > 155 Fleet Street, Portsmouth, NH 03801
> > www.bac-test.com
> > LinkedIn: http://www.linkedin.com/in/edwardhague
> >
> > <
> https://github.com/bacnet-stack/bacnet-stack/issues/515#issuecomment-1902668520>,
>
> > <