bacnet-stack/bacnet-stack #520: tcdrain blocks the thread

philippejohnrichard, 2023-10-24:
In the rs458.c
RS485_Send_Frame()

The line : 
tcdrain(RS485_Handle);

Freeze my listener thread after a while. 

Is there a safer way to :
/* wait until all output has been transmitted. */

or

Can I clear this error from my sender thread.

Note : RS485 handle value is 27 at start of program. Value stay the same after the bug crash my program.  (I'm using a module with ttySC0 on linux.)

philippejohnrichard, 2023-10-24:
Note :

Program crash alot faster when i use wireshark on  USB0, SC1 at the same time as i run my code on the SC0 port

skarg, 2023-10-30:
There is a note "Any attempts to use tcdrain() from a process which is a member of a background process group on a fildes associated with its controlling terminal, shall cause the process group to be sent a SIGTTOU signal. If the calling process is blocking or ignoring SIGTTOU signals, the process shall be allowed to perform the operation, and no signal is sent."
https://pubs.opengroup.org/onlinepubs/009696799/functions/tcdrain.html

Perhaps there needs to be a signal catcher to handle the SIGTTOU?