bacnet-stack/bacnet-stack #524: Cov not working

cyberumut, 2023-10-26:
Hello all, 

i'm using bacnet stack 0.8.5 version, i get when subscribe yabe to bacnet point always return Failed Subscribe error . 

how can i solve ?

skarg, 2023-10-26:
Has COV been enabled in your device for the particular object instance you are trying to SubscribeCOV?  COV for an object results in the server sending COVnotification service with the Present_Value and Status_Flags property values from a specific object instance a) upon subscription and b) any time present-value or status-flags change.

To enable COV for an object in the BACnet stack library in v0.8.x for a particular product:

1.  add extra files to your build: cov.c, h_cov.c, h_ccov.c, and there might be others (but the linker will warn if the functions are

missing).

2.  reference demo/object/bi.c - three functions are used by device.c module to a) detect when Present_Value or Status_Flags values change, and b) encode the present-value and status-flags property values to send in the COVnotification.

```
bool Binary_Input_Change_Of_Value(
    uint32_t object_instance);
void Binary_Input_Change_Of_Value_Clear(
    uint32_t object_instance);
bool Binary_Input_Encode_Value_List(
    uint32_t object_instance,
    BACNET_PROPERTY_VALUE * value_list);
```

If there are other object types that your device needs to support for COV, reference the master branch of the 1.x BACnet stack for object examples - they should be similar - and the encoding for binary has been refactored into bacnet/cov.c module.  See  `cov_data_value_list_link()`, `cov_value_list_encode_real()`, `cov_value_list_encode_enumerated()`, and `cov_value_list_encode_unsigned()`.

These unique object functions - for each object that supports COV - are added to the device.c module in the Object_Table.

For Analog objects, additional requirement of the COV Increment property value which is normally non-volatile.

```
    float Analog_Input_COV_Increment(
        uint32_t instance);
    void Analog_Input_COV_Increment_Set(
        uint32_t instance,
        float value);
```
3. demo/server/main.c - an apdu handler is configured to handle COV subscription.

4.  add to your APDU init: apdu_set_confirmed_handler(SERVICE_CONFIRMED_SUBSCRIBE_COV,  handler_cov_subscribe);  This handles subscriptions to the object instances.

5. demo/handler/h_cov.c - contains handler_cov_task() that is called in main loop manages the sending of the COV notification when objects indicate that their present-value or status-flags have changed.

6. demo/handler/h_cov.c - contains handler_cov_timer_seconds() that is called every second to manage the subscription timeouts.

7. The maximum number COV subscriptions are configured by these defines (in h_cov.c module):
```
#define MAX_COV_SUBCRIPTIONS 128
#define MAX_COV_ADDRESSES 16
```
8. The h_ccov.c file also needs at least one TSM (transaction state machine) buffer, defined in the Makefile or project precompiler defines as `MAX_TSM_TRANSACTIONS=1` or some other value larger than 0.

cyberumut, 2023-10-27:
thanks for answer Steve

closed 2023-10-27 by cyberumut