bacnet-stack/bacnet-stack #528: Doesn't compile on Windows with Code::Blocks without modifying CMakeLists.txt by hand

jd-imi, 2023-11-06:
If I try to follow instructions in README.md, i.e.:

```
c:\> mkdir build
c:\> cd build/
c:\> cmake .. -G"CodeBlocks - MinGW Makefiles"
```

I get :

```
C:\PROGRA~1\MINGW-~1\X86_64~1.0-P\mingw64\bin\gcc.exe -DBACDL_BIP -DBACNET_PROPERTY_LISTS -DBACNET_PROTOCOL_REVISION=19 -DBACNET_STACK_DEPRECATED_DISABLE -DBACNET_STACK_STATIC_DEFINE -DBAC_ROUTING -DPRINT_ENABLED=1 @CMakeFiles/bacnet-stack.dir/includes_C.rsp -O3 -DNDEBUG   /W3 /D_CRT_SECURE_NO_WARNINGS /wd4244 /wd4018 /wd4267 -o CMakeFiles\bacnet-stack.dir\src\bacnet\abort.c.obj -c D:\Documents\dev\BACnet_Stack_original_repo\bacnet-stack\src\bacnet\abort.c
gcc.exe: error: /W3: No such file or directory
gcc.exe: error: /D_CRT_SECURE_NO_WARNINGS: No such file or directory
gcc.exe: error: /wd4244: No such file or directory
gcc.exe: error: /wd4018: No such file or directory
gcc.exe: error: /wd4267: No such file or directory
```

If I apply this modification:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 4b9307dc..3676bcfa 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -538,7 +538,7 @@ elseif(WIN32)
     $<$<BOOL:${BACDL_BIP}>:ws2_32>
     $<$<BOOL:${BACDL_BIP}>:iphlpapi>)

-  add_definitions("/W3 /D_CRT_SECURE_NO_WARNINGS /wd4244 /wd4018 /wd4267")
+  #add_definitions("/W3 /D_CRT_SECURE_NO_WARNINGS /wd4244 /wd4018 /wd4267")

   target_sources(${PROJECT_NAME} PRIVATE
     ports/win32/bacport.h
```

It works

skarg, 2023-11-12:
Researching methods to detect specific compilers:
* [CMake: OS specific instructions](https://stackoverflow.com/questions/9160335/os-specific-instructions-in-cmake-how-to)
* [CMake: Setting build variables](https://cmake.org/cmake/help/latest/guide/user-interaction/index.html#setting-build-variables)
* [CMake: How to write platform checks](https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/How-To-Write-Platform-Checks)

The problem line is MSVC specific, so we should wrap a check around that:
```
if(MSVC)
    add_definitions("/W3 /D_CRT_SECURE_NO_WARNINGS /wd4244 /wd4018 /wd4267")
endif()
````

Does that work for you?

jd-imi, 2023-11-13:
It is working well with the change you proposed.

PR #533 (2023-11-13): fix CMake for code::blocks build
fixed #528 by adding MSVC checking around MSVC specific options to allow CMake to build successfully on Windows with code::blocks tool option.
closed 2023-11-13 by skarg