bacnet-stack/bacnet-stack #542: Read ExceptionSchedule

IgorIIV, 2023-11-30:
I am trying to read the ReadPropertyMultiple command of the ExceptionSchedule property in Schedule. Using an example from the stack delivery. The code goes all the way to the method:
`BACNET_APPLICATION_TAG method bacapp_context_tag_type(BACNET_PROPERTY_ID property, uint8_t tag_number)`

tag_number is 0. As a result, the CASE exits and gives the status BACNET_STATUS_ERROR:
```
        case PROP_EXCEPTION_SCHEDULE:
            switch (tag_number) {
                case 1:
                    tag = BACNET_APPLICATION_TAG_OBJECT_ID;
                    break;
                case 3:
                    tag = BACNET_APPLICATION_TAG_UNSIGNED_INT;
                    break;
                **case 0: /* calendarEntry: abstract syntax + context */**
                case 2: /* list of BACnetTimeValue: abstract syntax */
                default:
                    break;
            }
            break;
```

Does reading ReadPropertyMultiple for the ExceptionSchedule property of the Schedule object work?
I attach a link to the Wireshark log.
[Schedule.zip](https://github.com/bacnet-stack/bacnet-stack/files/13512508/Schedule.zip)
As a device I use Bacnet Device Simulator
https://www.scadaengine.com/software6.html

MightyPork, 2023-11-30:
Hi, yes that is implemented and works, but I didn't get around to merge it yet.

See https://github.com/bacnet-stack/bacnet-stack/pull/474

Everything relating to Schedule and Calendar should be supported on that branch - if you don't mind it's a few months stale, you can try it

MightyPork, 2023-11-30:
Yes, writing works. I'm not sure if it's implemented in the example apps that come with bacnet stack though - I am using it as a statically linked library and call the API directly.

there is no schedule or plan for merging, it will happen, but I didn't have time for that yet

closed 2024-02-05 by skarg