bacnet-stack/bacnet-stack #558: Slow response when reading and writing attributes

Twillpower, 2024-01-18:
I use bacnet linux to write(and read) attributes to esp32 bacnet through mstp. I found that sometimes the response is very fast, and sometimes it is very slow. Through the logic analyzer and mstpcap, I found that it is due to the large delay in sending the write(and read) request frame by bacnet linux. This is the accumulation of data. Into the queue? But by calling Ringbuf_Count, I found that there is no large amount of data accumulated in the queue. What are the possible reasons for this?

skarg, 2024-01-18:
The MSTP capture would help determine the problem.  The likely culprit is timing. Typical Linux (or Windows) running the MSTP process does not keep consistent timing that has less than 5ms of jitter. 

There is a note in the C file ports/linux/rs485.c that suggests:
```
/* Use ionice wrapper to improve serial performance:
   $ sudo ionice -c 1 -n 0 ./bin/bacserv 12345
*/
```

Alternately, a Linux based BACnet MSTP Line Discipline Driver could be used to get the real time process MSTP is expecting.

Twillpower, 2024-01-20:
sudo ionice -c 1 -n 0 nice -n -20 ./S_P_S_BACNET_LINUX 12345
I have tried to adjust the priority to the highest, but there is still a situation like this, and I have logged in some key places.
```
```

Twillpower, 2024-01-20:
```
void wy_read_prop(wy_bacnet_rw_data_t * p_wy_bacnet_rw_data)
{
    unsigned max_apdu        = MAX_PDU;
    bool     found           = false;
    time_t   last_seconds    = 0;
    time_t   current_seconds = 0;
    Error_Detected = false;
    uint8_t * mac;
    if (address_get_by_device(p_wy_bacnet_rw_data->device_id, &max_apdu, &Target_Address))
    {
        found = true;
    }
    if (found)
    {
        last_seconds = time(NULL);
        do
        {
            current_seconds = time(NULL);
            if (current_seconds != last_seconds)
            {
                tsm_timer_milliseconds(
                    (uint16_t)((current_seconds - last_seconds) * 1000));
            }
            if (Request_Invoke_ID == 0)
            {
                Request_Invoke_ID = Send_Read_Property_Request_Address(&Target_Address, MAX_PDU,
                                                                       p_wy_bacnet_rw_data->object_type, p_wy_bacnet_rw_data->object_instance, p_wy_bacnet_rw_data->object_property, BACNET_ARRAY_ALL);
                printf("Send Read Prop Invoke: %d", Request_Invoke_ID);
            }
            else if (tsm_invoke_id_free(Request_Invoke_ID))
            {
                Request_Invoke_ID = 0;
            }
            else if (tsm_invoke_id_failed(Request_Invoke_ID))
            {
                printf("Error: Read Timeout!\n");
                tsm_free_invoke_id(Request_Invoke_ID);
                Request_Invoke_ID = 0;
                Error_Detected    = true;
            }
            if (Error_Detected)
            {
                mac = wy_get_mac_by_dev_id(p_wy_bacnet_rw_data->device_id);
                if (mac != NULL)
                {
                    wy_report_to_mqtt_press_state(mac, false);
                }
                break;
            }
            usleep(10 * 1000);
            last_seconds = current_seconds;
        } while (Request_Invoke_ID != 0);
    }
}

void wy_write_prop(wy_bacnet_rw_data_t * p_wy_bacnet_rw_data)
{
    unsigned max_apdu        = 0;
    bool     found           = false;
    time_t   last_seconds    = 0;
    time_t   current_seconds = 0;
    Error_Detected = false;
    uint8_t * mac;
    if (address_get_by_device(p_wy_bacnet_rw_data->device_id, &max_apdu, &Target_Address))
    {
        found = true;
    }
    if (found)
    {
[... 189 lines cut ...]
object_type: WY_BAC_OBJ_CURTAIN object_instance: 0 object_property: WY_PROP_PERCENT
 current value: 100
2022-05-03 11:32:29.802
2022-05-03 11:32:29.803
2022-05-03 11:32:29.803
2022-05-03 11:32:29.803
2022-05-03 11:32:29.803
2022-05-03 11:32:29.804
2022-05-03 11:32:29.804
2022-05-03 11:32:29.804
2022-05-03 11:32:29.804
2022-05-03 11:32:29.805
2022-05-03 11:32:29.805
2022-05-03 11:32:29.805
2022-05-03 11:32:29.805
2022-05-03 11:32:29.806
2022-05-03 11:32:29.806
2022-05-03 11:32:29.806
2022-05-03 11:32:29.806
2022-05-03 11:32:29.807
2022-05-03 11:32:29.812
2022-05-03 11:32:29.812
2022-05-03 11:32:29.813
read after 2022-05-03 11:32:29.813
```

Twillpower, 2024-01-20:
Thank you sir, the problem has been solved after debugging

closed 2024-01-20 by skarg