bacnet-stack/bacnet-stack #567: Problem with Analog_Input_Present_Value_Set

GauiStori, 2024-02-04:
Hi

I do have a very strange problem with the bacserv program.

I added a few lines to make one of the analog-inputs change values during runtime,
see patch below.

Then I get the erroneous output:

BACnet Device Name: SimpleServer
Setting value 0 to 0.000000
Inside function: instance = 0 value = 0.000000
Setting value 1 to 10.000000
Inside function: instance = 1 value = 0.000000
Setting value 0 to 0.100000
Inside function: instance = 0 value = -0.000000
Setting value 1 to 10.100000
Inside function: instance = 1 value = 2.000000
Setting value 0 to 0.200000
Inside function: instance = 0 value = -0.000000
Setting value 1 to 10.200000
Inside function: instance = 1 value = 36893488147419103232.000000

I have tried with gcc-11,12,13 and clang-16 but I get the same error for all compilers.

But if I change the function definition to using double I get the following correct output.
Using int32_t works as well.

BACnet Device Name: SimpleServer
Setting value 0 to 0.000000
Inside function: instance = 0 value = 0.000000
Setting value 1 to 10.000000
Inside function: instance = 1 value = 10.000000
Setting value 0 to 0.100000
Inside function: instance = 0 value = 0.100000
Setting value 1 to 10.100000
Inside function: instance = 1 value = 10.100000
Setting value 0 to 0.200000
Inside function: instance = 0 value = 0.200000
Setting value 1 to 10.200000
Inside function: instance = 1 value = 10.200000

[value_change.txt](https://github.com/bacnet-stack/bacnet-stack/files/14158606/value_change.txt)

Thanks and regards
Gudjon

GauiStori, 2024-02-05:
Hi again

I got a hint that this might be some kind of a stack corruption and it works because double is larger than float.

Now I changed the code to using int32_t.

Incrementing val with 1 at a time works.
Incrementing val with 0x10000000 at a time works also perfectly

BACnet Device Name: SimpleServer
Setting value 0 to 0
Inside function: instance = 0 value = 0
Setting value 1 to A
Inside function: instance = 1 value = A
Setting value 0 to 10000000
Inside function: instance = 0 value = 10000000
Setting value 1 to 1000000A
Inside function: instance = 1 value = 1000000A
Setting value 0 to 20000000
Inside function: instance = 0 value = 20000000
Setting value 1 to 2000000A
Inside function: instance = 1 value = 2000000A
Setting value 0 to 30000000
Inside function: instance = 0 value = 30000000
Setting value 1 to 3000000A
Inside function: instance = 1 value = 3000000A

Regards
Gudjon

skarg, 2024-02-05:
Perhaps you did not include `Analog_Input_Present_Value_Set()` function prototype in apps/server/main.c:
```
#include "bacnet/basic/object/ai.h"
```
When a function is not declared in a compilation unit, the C language specifies that the following implicit declaration is made:
```
int f();
```
which is a function accepting any number of parameters and returning an `int`.

The following changes increment the present-value of Analog Input 0 by 10.0 every second when added to the server example:
```
...
/* include the device object */
#include "bacnet/basic/object/device.h"
#include "bacnet/basic/object/ai.h"
...
int main(int argc, char *argv[])
{
...
    float ai_present_value = 0.0F;
...
        if (mstimer_expired(&BACnet_Task_Timer)) {
            mstimer_reset(&BACnet_Task_Timer);
...
            Analog_Input_Present_Value_Set(
                Analog_Input_Index_To_Instance(0),
                ai_present_value);
            ai_present_value += 10.0F;
```

GauiStori, 2024-02-06:
Thanks a lot Steven for this and the library

Sorry. My stupid mistake. I didn't know C behaved like that :)

Regards
Gudjon

closed 2024-02-06 by GauiStori