bacnet-stack/bacnet-stack #571: segmentation support & BTL Cert

cyberumut, 2024-02-09:
When will segmentation be supported?
And does bacnet stack pass BTL Certification  ?

what i need for BTL Certification ?

cyberumut, 2024-02-09:
where to i look for segmentation ? i want add segmentation for bacnet stack lib

MightyPork, 2024-02-09:
unfortunately not supported yet. there was some work but nothing finished.
this is a duplicate of #362 #417

closed 2024-02-09 by MightyPork

skarg, 2024-02-12:
BTL testing and certification information is available from the https://btl.org/ website.

This BACnet Stack protocol library has been used in many products as the basis for their BACnet functionality, and those products have passed BTL testing.  Any defects found in this library via BTL testing are corrected.  However, a lot of the BACnet functionality depends on how the vendor modeled their device, the objects and properties and services and datalinks they chose to use, and a variety of other variables.