bacnet-stack/bacnet-stack #576: Zephyr hello_bacnet_stack sample build fail

gp-src, 2024-02-20:
The bacnet-stack\zephyr\samples\hello_bacnet_stack sample fails to build out the box with the warning:

_warning: attempt to assign the value 'y' to the undefined symbol BACNETSTACK_

I had to add a Kconfig file within the hello_bacnet_stack project folder to build OK:
```
rsource "../../Kconfig"

source "${ZEPHYR_BASE}/Kconfig.zephyr"
```

legrand-gregshue, 2024-02-22:
@gp-src Thank you for letting us know. What manifest file are you using? What is the version of Zephyr you are building against? What was the build command line that failed?

gp-src, 2024-02-23:
Thanks for getting back to me @legrand-gregshue.

My west.yml manifest file:
```

manifest:
  version: 0.13

  remotes:
    - name: bacnet-stack
      url-base: https://github.com/bacnet-stack
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos

  defaults:
    remote: bacnet-stack

  self:
    path: bacnet-stack

  projects:
    - name: zephyr
      path: zephyr
      revision: v3.4.0
      remote: zephyrproject-rtos
      west-commands: scripts/west-commands.yml
      import:
        name-allowlist:
        - cmsis
        - picolibc
    - name: hal_st
      revision: 5948f7b3304f1628a45ee928cd607619a7f53bbb
      path: modules/hal/st
      remote: zephyrproject-rtos
      groups:
        - hal
    - name: hal_stm32
      revision: c865374fc83d93416c0f380e6310368ff55d6ce2
      path: modules/hal/stm32
      remote: zephyrproject-rtos
      groups:
        - hal
```

Zephyr version 3.4.

Build command used:
`west build -b nucleo_h743zi -p`

Adding the below line to the hello_bacnet_stack CMakeLists.txt file also built the sample OK.
`set(ZEPHYR_EXTRA_MODULES ../../../)`

So I assume I needed to set this in my environment variables to point to the bacnet-stack/zephyr/ path?

gp-src, 2024-02-24:
I think I know where I went wrong then... I had ZEPHYR_BASE set in my environment variables to point to the Zephyr 3.4.0 path, so instead of using the west.yml in bacnet-stack, it was using the west.yml in the zephyr directory, which obviosuly didn't contain any reference to the bacnet-stack.

I am hoping to get to try this library properly soon. I do have a seperate query about the static memory usage expected for the library... but i'll open a seperate discussion thread for that. 

Thanks for the info again!

closed 2024-02-24 by gp-src