bacnet-stack/bacnet-stack #590: undefined reference to 'xxx_function' with cgo_gcc when build the golang codes with shared library

badboy-huaqiao, 2024-03-05:
Hi, @skarg , I'm planning to make a wrapper with golang cgo，I had get the static library libbacnet.a with the command `make lib` , but when I build the golang codes, I get the errors which can't find some functions' reference. so whether the function in the examples is already exported in the static library, such as `Device_Init`. the error message is as follow:

```
in function `_cgo_fe2d2f535939_Cfunc_Device_Init': /tmp/go-build/cgo-gcc-prolog:51: undefined reference to `Device_Init'
libbacnet.a(h_rp.o):h_rp.c:(.text$handler_read_property+0x117): undefined reference to `Device_Object_Instance_Number'
libbacnet.a(h_rp.o):h_rp.c:(.text$handler_read_property+0x133): undefined reference to `Network_Port_Index_To_Instance'
libbacnet.a(h_rp.o):h_rp.c:(.text$handler_read_property+0x170): undefined reference to `Device_Read_Property'
libbacnet.a(s_iam.o):s_iam.c:(.text$iam_encode_pdu+0x47): undefined reference to `Device_Vendor_Identifier'
```

badboy-huaqiao, 2024-03-05:
I extract the functions list of libbacnet.a, and I find there is indeed no relevant functions in the file, such as `Device_Read_Property` is not in the` h_rp.c` file.
[image]

skarg, 2024-03-06:
The apps/lib/Makefile does not include a device object or its children objects.  The example apps built with Makefile add bacnet/basic/object/device.c or bacnet/basic/object/client/device-client.c, and net-port.c files to build their functionality.

```
# BACnet objects that are used with this app
BACNET_OBJECT_DIR = $(BACNET_SRC_DIR)/bacnet/basic/object
SRC = main.c \
	$(BACNET_OBJECT_DIR)/client/device-client.c \
	$(BACNET_OBJECT_DIR)/netport.c
```

I am in the process of adding a bacnet/basic/services/h_device.c module that will remove the library dependency on device.c module, but a device object is still required (even for BACnet clients).

badboy-huaqiao, 2024-03-07:
really thanks for your reply, so you mean that the static library I made with `make lib` is not the BACnet library, it's just the apps examples static library, do I make sense?

badboy-huaqiao, 2024-03-07:
thanks for your reply, so you mean that the static library I made with` make lib` is not the BACnet library, it's just the apps examples static library, so I can't make a golang client wrapper with that static library, do I make sense?
actually, I just want to make a golang client wrapper that can read and write the BACnet device, the server features are not included.
I have made a golang wrapper codes with reference to your sample codes `apps/readprop/main.c`，but I got errors  mentioned  above. 
so can you make a new simple example showing how to read and write the BACnet device, just client features, so I can create a golang wrapper based on your sample codes, if that works, I will contribute the golang wrapper to your repository, so the golang engineers can easily use this great C stack.
great thanks for your efforts.

skarg, 2024-05-03:
You could make an example that doesn't include any server features (not BACnet compliant) - just remove the server handlers in the apps/readprop/main.c example that make the device a server:
```
    /* we need to handle who-is
       to support dynamic device binding to us */
    apdu_set_unconfirmed_handler(SERVICE_UNCONFIRMED_WHO_IS, handler_who_is);

    /* we must implement read property - it's required! */
    apdu_set_confirmed_handler(
        SERVICE_CONFIRMED_READ_PROPERTY, handler_read_property);
```