bacnet-stack/bacnet-stack #616: the bacserv of  MSTP in win32 work abnormal !

ccc0818, 2024-04-15:
the bacnet-stack version is 1.0.0

my bacserv work on windows11 amd64, the compiler is i686_mingw32_sjij

the bacrp work on linux arm64, 

The bacserv Linux version compiled with the same code can run normally, but the win32 version Only a very small probability of occasionally reading data once or twice 

[image]
[image]

skarg, 2024-05-01:
If I may quote from [Adafruit](https://learn.adafruit.com/blinking-led-with-zephyr-rtos/next-steps): 

> User-facing operating systems like Windows, OSX and Linux are not deterministic, allotting time to tasks like memory management or storage at any time, meaning they often experience delays or lags in operation.

MS/TP can only tolerate delays of - at most - 5ms, and user-facing operating systems, in particular Windows, can rarely provide the real-time performance needed by MS/TP.

ccc0818, 2024-05-01:
thx,
I roughly understand, but why is there still a win32 port provided here？

skarg, 2024-05-02:
Why is there still a Windows MS/TP port? It's always possible that someone takes an interest and wants to fix or improve the Windows MS/TP port to work better or properly.

When I wrote the win32 MSTP port, it worked somewhat (Windows 95?). It's also building with MinGW by default, and perhaps MSVC can build it better/faster. There is also an alternate version using the multimedia timer [ports/win32/dlmstp-mm.c](https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/win32/dlmstp-mm.c).   I've read that [αcτµαlly pδrταblε εxεcµταblε](https://justine.lol/ape.html) improves the execution speed of POSIX applications on Windows, but haven't tried building using that method.

ccc0818, 2024-05-02:
thx again, i'll try it.

closed 2024-05-02 by ccc0818