bacnet-stack/bacnet-stack #621: Issue with writing schedule using /apps/writeprop/bacwp

Girgitt, 2024-04-18:
Hello, Is there a set of examples how to write certain objects using demo application bacwp? In specific the schedule object. 

Platform: amd64, 64bit
OS: Rocky Linux release 8.6 (Green Obsidian)
bacnet-stack version: tag 1.3.4

compiled with: 
```
# make clean
# make build all
```
The bacwp application seems to work ok:
```
# /tmp/bacnet-stack/apps/readprop/bacrp 99 17 22334455 123
(Unsigned Int; Mon: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Tue: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Wed: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Thu: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Fri: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sat: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sun: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3])

```
However, when I try to write the schedule back I get error:
```
# /tmp/bacnet-stack/apps/writeprop/bacwp 99 17 22334455 123 8 -1 15 '(Unsigned Int; Mon: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Tue: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Wed: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Thu: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Fri: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sat: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sun: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3])'
BACnet Error: property: value-out-of-range
```

Girgitt, 2024-04-18:
ok I think the issue is with "Tag" value. in /src/bacnet/bacenum.h BACNET_APPLICATION_TAG_WEEKLY_SCHEDULE is enum entry 30 and I was passing 15 (without checking the meaning to be honest but using 2 or 4 did not change anything).

In comments to #278 I found (https://github.com/bacnet-stack/bacnet-stack/issues/278#issuecomment-1208246736) the prefered way is to pass -1 as tag and voila! write worked: 

```
$ ./bacwp 99 17 22334455 123 8 -1 -1 '(Unsigned Int; Mon: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Tue: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Wed: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Thu: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Fri: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sat: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sun: [00:00:00.00 1, 08:00:00.00 2, 18:00:00.00 3])'

WriteProperty Acknowledged!
$ ../bacrp 16 99 22334455 123
(Unsigned Int; Mon: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Tue: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Wed: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Thu: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Fri: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sat: [00:00:00.00 3, 08:00:00.00 3, 18:00:00.00 3]; Sun: [00:00:00.00 1, 08:00:00.00 2, 18:00:00.00 3])

```

Girgitt, 2024-04-18:
My issue is resolved and this ticket can be closed unless someone can comment on documentation/resources covering more examples of writing values for different types of object.

skarg, 2024-04-27:
I used the property along with "-1" to decode the command line parameter for complex or constructed datatypes. I'll add a note about the usage of tag = -1 in the WriteProperty example app help message.

closed 2024-04-27 by skarg