bacnet-stack/bacnet-stack #671: The MSTP device object does not have the object_lists attribute

fangjj0709, 2024-06-12:
When I was using bacnet stack version 1.3.5, I found that when creating an MSTP device, the object_list attribute was not present under the device object, but it was present under the IP device creation. Even with readprop, I couldn't read this attribute. Why is this?
[image]

skarg, 2024-06-12:
v1.3.5 enabled MS/TP extended frames, by default, for APDU up to 1476 bytes [#592].  It's possible that it was implemented wrong although our testing showed otherwise.  It's also possible that the router or client you are testing with is not MS/TP extended frames compatible.  A Wireshark capture of the MS/TP might be informative.

fangjj0709, 2024-06-12:
> v1.3.5 默认启用了 MS/TP 扩展帧，APDU 最大可达 1476 字节 [ #592 ]。虽然我们的测试结果并非如此，但这可能是错误实现的。您测试的路由器或客户端也可能不兼容 MS/TP 扩展帧。Wireshark 捕获的 MS/TP 可能会提供有用信息。

Thank you for your reply. The readprop I am using here is also compiled from version 1.3.5. Here is my printed information
`NPDU: BACnet Protocol Version=87.  Discarded!`
May I ask which version does not enable this MS/TP extension frame? Can switching versions solve this problem.

skarg, 2024-06-12:
Please change MAX_APDU=480 (in config.h file or in Makefile or ...) to see if that makes a difference.  The extended frames only happens when the packet exceeds 480 bytes (and object-list is often larger than 480 bytes with a lot of objects).

Extended frames was introduced in v1.3.2 (although it was broken).

fangjj0709, 2024-06-12:
> 请更改 MAX_APDU=480（在 config.h 文件或 Makefile 中或...）以查看是否有区别。仅当数据包超过 480 字节时才会发生扩展帧（并且对象列表通常大于 480 字节，并且包含大量对象）。
> 
> 扩展框架在 v1.3.2 中被引入（尽管它已被破坏）。

I tried setting MAX-APDU to 480, and there will be the following printing information
`BACnet Abort: Segmentation Not Supported`
But I'm not sure how to set this Segmentation. Can you ask if there is any way to set this property,
Then I reduced the number of objects and reduced the number of packets to no more than 480 to obtain the object_list correctly.

skarg, 2024-06-12:
The current BACnet Stack library does not support segmentation, and segmentation is not needed or required for a BACnet server or client device.

See the [BACnet Testing Laboratories - Implementation Guidelines](https://btl.org/wp-content/uploads/sites/3/2022/08/btl_implementation_guideline-1.pdf)

> 5.6 Devices supporting segmentation shall be able to use non-segmented methods to retrieve large
amounts of data
> Client devices that support segmentation and retrieve large amounts of data shall be prepared to use nonsegmented services when server devices indicate they do not support segmentation.
> 
> For example, the Object_List for a device may not fit into a single segment. In this case, the individual array
items shall be read. One method is to read the size of the Object_List property and then read each item
individually using the ReadProperty service or a group of items could be read using ReadPropertyMultiple
service provided that each request and answer response can be placed in a single segment.

There is an example BACnet client application and library module that reads the object-list.  See apps/server-client for the example app, and src/bacnet/basic/client for the library module(s).

closed 2024-07-02 by skarg