bacnet-stack/bacnet-stack #706: bacnet-stack building on linux

kush930, 2024-07-29:
Hi all, 

`I cloned master branch bacnet-stack from link`: https://github.com/bacnet-stack/bacnet-stack.git

**I executed command:-sudo make clean all**

`I am facing issue after running command: sudo make clean all
`
`Issue:-`

```
gcc -c -Wall -I. -g -O2 dbuffer.c
gcc -c -Wall -I. -g -O2 bigendian.c
gcc -c -Wall -I. -g -O2 version.c
gcc -o bacnet4linux main.o options.o net.o bacnet_text.o invoke_id.o bacdcode.o html.o return_properties.o signal_handler.o check_online_status.o query_new_device.o bacnet_object.o bacnet_device.o send_npdu.o send_read_property.o send_write_property.o send_subscribe_cov.o send_whois.o send_iam.o send_time_synch.o send_bip.o packet.o ethernet.o receive_apdu.o receive_readproperty.o receive_npdu.o receive_readpropertyACK.o receive_COV.o receive_iam.o receive_bip.o debug.o pdu.o reject.o keylist.o dstring.o dbuffer.o bigendian.o version.o 
/usr/bin/ld: invoke_id.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: html.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: signal_handler.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: check_online_status.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: query_new_device.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: send_npdu.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: send_subscribe_cov.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: send_bip.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: ethernet.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: receive_npdu.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: receive_iam.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: receive_bip.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
/usr/bin/ld: pdu.o:/home/kush/Downloads/bacnet/main.h:68: multiple definition of `deviceRangeList'; main.o:/home/kush/Downloads/bacnet/main.c:85: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:29: bacnet4linux] Error 1
```

if anyone has solution, please help me, where i am doing mistake

T&R
kush

skarg, 2024-07-29:
Maybe you have the bacnet4linux repository instead of the bacnet-stack repository? They are separate projects, and we haven't updated bacnet4linux in many years.

teksturi, 2024-08-15:
Did you manage to solver your build problem?