bacnet-stack/bacnet-stack #752: Possibly leak when setting the object name

imari-globant, 2024-08-27:
If you create a child object using the Device_Create_Object function, there is no way of setting the name from there. So you could use a function for the particular object type e.g. Analog_Value_Name_Set, passing as a parameter a char*, where you previously allocated some memory for the string.
If you want to remove that object later, there is no way to get that pointer to free it. This is because Analog_Value_Name is not exported. So you cannot call it from your application code.
Looking at Analog_Value_Delete and Analog_Value_Cleanup, those functions do not free any pointer present in the name field.

skarg, 2024-08-27:
Thank you for reporting the problem. I didn't add any 'free' for Object_Name in the basic dynamic objects since I typically create the name statically from a table, rather than using malloc/calloc.

I see in av.h file that there is already an API to get the name pointer to be able to free it if it was created dynamically, and probably should be implemented in the basic dynamic objects to enable them to be freed. It is also missing the 'BACNET_STACK_EXPORT'
```
    char *Analog_Value_Name(
        uint32_t object_instance);
```

imari-globant, 2024-08-27:
Besides this fix. Would you think it would be worth having an additional field in BACNET_CREATE_OBJECT_DATA, so that we could create the object with a name? If that pointer is null, it could ignore it and work as currently implemented.

skarg, 2024-08-27:
The BACNET_CREATE_OBJECT_DATA already includes a 'List of Initial Values', but that feature is not currently supported in the basic device service handler for CreateObject service:
```
            if (data->list_of_initial_values) {
                /* FIXME: add support for writing to list of initial values */
                /*  A property specified by the Property_Identifier in the
                    List of Initial Values does not support initialization
                    during the CreateObject service. */
                data->first_failed_element_number = 1;
                data->error_class = ERROR_CLASS_PROPERTY;
                data->error_code = ERROR_CODE_WRITE_ACCESS_DENIED;
                /* and the object shall not be created */
```

The list_of_initial_values might be better decoded similar to the WriteProperty application data (data blob versus linked list of BACNET_PROPERTY_VALUE) as it would certainly take less memory and not have to worry about how many are needed before decoding.

```
List of Initial Values
This parameter shall convey a list of BACnetPropertyValues that shall be used to initialize the values of the specified properties
of the newly created object.
```
I would expect that the Device_Create_Object List of Initial Values handler would call into the Device Object WriteProperty handler to write each property value that is being initialized. In that case, each basic object WriteProperty would need to support writing to the object-name property and dynamically creating the new name and free it at the cleanup.