bacnet-stack/bacnet-stack #768: stm32f4xx demo project can't run

joanna99, 2024-09-20:
I used the demo project to download to NUCLEO-F429ZI board. The RS485 communication is ok.  I used PC yabe to add the board. But it  
 show can't get object list.

joanna99, 2024-09-20:
[image]

skarg, 2024-09-29:
I tested today the ports/stm32f4xx project built with Cmake and YABE is able to discover. I am connected via a BAS-Router (BACnet/IP to MS/TP router), not directly.

Can you check the MAX_APDU in the build?  For CMake:
```
target_compile_definitions(${EXECUTABLE} PRIVATE
    -DNDEBUG
    -DUSE_STDPERIPH_DRIVER
    -DSTM32F4XX
    -DBACDL_MSTP
    -DMAX_APDU=480
    -DBIG_ENDIAN=0
    -DMAX_TSM_TRANSACTIONS=1
    -DBACAPP_MINIMAL
)
```
For Makefile:
```
BACNET_FLAGS += -DMAX_APDU=480
```
If the value is larger than 480, it is possible that the device 'appears' to not respond as the MS/TP extended frame types are not yet supported by Yabe (and some routers).

jd-imi, 2024-12-19:
I had a similar issue with YABE with a MSTP device which doens't support segmentation:

YABE 1.2.2 is OK
YABE 1.2.4, 1.3.1 and 1.3.2 not OK

I report the issue to YABE project page but after some time, the content of the report disappear (but there is still the title with the link) :
https://sourceforge.net/p/yetanotherbacnetexplorer/discussion/help/thread/778d201dbf/

The fix is there :  https://sourceforge.net/p/yetanotherbacnetexplorer/code/467/

Still no release, but a binary with the fix is available:
https://sourceforge.net/p/yetanotherbacnetexplorer/code/HEAD/tree/trunk/Yabe/latestBin/

Hope this helps