bacnet-stack/bacnet-stack #776: Specify pin for DE/RE on a MAX485

deve87, 2024-09-23:
I wanted to test atmega168 port. Complied OK, but can't understand where to specify the output pins for RE/DE?
I apologize if it's obvious, but I'm not so familiar with C coding.

skarg, 2024-09-23:
Some things about the atmega168 port:

- I configured the project to use Atmega328 instead of Atmega168 since the code no longer fits into Atmega168 flash space using GCC compiler
- The code does very little - just exposing some I/O as Binary or Analog objects.
- The RS485 abstraction (code) is done in rs485.c module.

The RE/DE is accomplished in ports/atmega168/rs485.c module:
```
void RS485_Transmitter_Enable(bool enable)
{
    if (enable) {
        BIT_SET(PORTD, PD2);
    } else {
        BIT_CLEAR(PORTD, PD2);
    }
}
```

deve87, 2024-09-23:
Thank you.

I connected the RE/DE to PD2 and seem to work. But can't get the atmega328 to communicate with an Johnson Controls NAE on the MSTP port. The LED's show communicate between, but no device is discovered in J.C.

The correct thing is to change the atmega328 clock speed in hardware.h to 16mhz (F_CPU 16000000UL) and change "RS485_Baud" in rs485.c to 38400 right? Or do I miss something here?
I use an Arduino Uno with the ATmega328 and MAX485 transceiver

skarg, 2024-09-23:
Thanks for the updates when using the Arduino Uno board with ports/atmega168 example.
I developed the ports/atmega168 example for a board I designed using Atmega168, so that is why the values don't match Arduino Uno.

In ports/atmega168/main.c where the program starts, the baud is set to 38400 via RS485_Set_Baud_Rate(38400); function call. The RS485_Baud in rs485.c module is the default value if not set via the function.

Yes, there could be timing or other issue. I haven't tested with an Arduino Uno Atmega238 and RS485 transceiver yet, but it should be easy to do this.  The timing can be checked with oscilloscope or serial analyzer.

deve87, 2024-09-23:
Would be great to have a working port for the uno with MSTP some day. Have no experience on how to do so myself, checking for timing issues and so on.

Thanks again for you'r replies.

deve87, 2024-09-24:
The only problem was I had not set a valid MAC address. So when I connected the Uno to the Johnson Controls NAE with an MAC of 0. The J.C would not establish communication. Simply set MAC to 5 solved the problem. Now able to discover the Uno.

deve87, 2024-09-24:
> * The code does very little - just exposing some I/O as Binary or Analog objects.

Can I as if the BV or AV object is supposed to change with the atmega I/O pins?
If I change the DDRB = 0 to 1 All the PB port acts as outputs but is it possible to write PV and  then change the these pins to low/ high?

deve87, 2024-09-25:
Great! Will test this out when I have the time. 
Made some setups with bacpypes and raspberry GPIO over BACnetIP. Would be great using an Uno as well over MSTP with some relays and DI inputs using I/O.

Thanks for replying.

closed 2024-09-27 by skarg