bacnet-stack/bacnet-stack #778: router-ipv6 cmake compile error

stargieg, 2024-09-23:
if compile with cmake and BACDL_BIP=ON and BACDL_BIP6=ON

```
duplicate symbol '_Device_Object_Instance_Number'
```

I think the function can be removed, but then Makefile will no longer work.

skarg, 2024-09-23:
router-ipv6 was designed as a layer-2 router with no application layer (no device object) - cmake is adding a device object to the router-ipv6 build?
```
/**
 * @brief Get the Device object instance number
 * @return The Device object instance number
 * @note This is a proxy function to satisfy the BACnet Stack IPv6 port layer
 * requirements since this application omits a Device object.
 */
uint32_t Device_Object_Instance_Number(void)
{
    return Device_Instance_Number;
}
```

stargieg, 2024-09-24:
The Device_Object_Instance_Number come from libbacnet-stack.a
With Linux and Apple.
```
[build] [ 20%] Building C object CMakeFiles/bacnet-stack.dir/src/bacnet/basic/object/device.c.o
...
[build] [ 86%] Building C object CMakeFiles/router-ipv6.dir/apps/router-ipv6/main.c.o
...
[build] [ 91%] Linking C executable router-ipv6
...
[build] duplicate symbol '_Device_Object_Instance_Number' in:
[build]     /Users/patrick/bacnet-stack-upstream-1/build/CMakeFiles/router-ipv6.dir/apps/router-ipv6/main.c.o
[build]     libbacnet-stack.a[57](device.c.o)
[build] ld: 1 duplicate symbols
[build] clang: error: linker command failed with exit code 1 (use -v to see invocation)
[build] gmake[2]: *** [CMakeFiles/router-ipv6.dir/build.make:98: router-ipv6] Error 1
[build] gmake[1]: *** [CMakeFiles/Makefile2:785: CMakeFiles/router-ipv6.dir/all] Error 2
[build] gmake[1]: *** Waiting for unfinished jobs....
```
build.make
```
# External object files for target router-ipv6
router__ipv6_EXTERNAL_OBJECTS =

router-ipv6: CMakeFiles/router-ipv6.dir/apps/router-ipv6/main.c.o
router-ipv6: CMakeFiles/router-ipv6.dir/build.make
router-ipv6: libbacnet-stack.a
router-ipv6: CMakeFiles/router-ipv6.dir/link.txt
	@$(CMAKE_COMMAND) -E cmake_echo_color "--switch=$(COLOR)" --green --bold --progress-dir=/Users/patrick/bacnet-stack-upstream-1/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_2) "Linking C executable router-ipv6"
	$(CMAKE_COMMAND) -E cmake_link_script CMakeFiles/router-ipv6.dir/link.txt --verbose=$(VERBOSE)
```
link.txt
```
/usr/bin/clang -g -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names "CMakeFiles/router-ipv6.dir/apps/router-ipv6/main.c.o" -o router-ipv6  libbacnet-stack.a
```

commit a71d4b95 (2024-10-14): Fix/router ipv6 test (#806)

* fix app router-ipv6: duplicate symbol
Makefile use Device_Object_Instance_Number from device-client.c (Device_Object_Instance_Number is not in bacnet.a from Makefile)

Cmake use Device_Object_Instance_Number from libbacnet-stac.a:device.c

This fix #778

* test ipv6
PR #807 (2024-10-14): fix app router-ipv6: duplicate symbol
Makefile use Device_Object_Instance_Number from device-client.c (Device_Object_Instance_Number is not in bacnet.a from Makefile)

Cmake use Device_Object_Instance_Number from libbacnet-stac.a:device.c

This fix #778

With this patch we can turn BACDL_BIP6 on.
commit 2d98e2f6 (2024-10-14): fix app router-ipv6: duplicate symbol (#807)

Makefile use Device_Object_Instance_Number from device-client.c (Device_Object_Instance_Number is not in bacnet.a from Makefile)

Cmake use Device_Object_Instance_Number from libbacnet-stac.a:device.c

This fix #778
closed 2024-10-14 by skarg