bacnet-stack/bacnet-stack #782: Reading the PrensentValue of the bacnet/MSTP device timed out

fangjj0709, 2024-09-24:
When using version 1.3.8 of the repository, it was found that when using the readProperty service to read the present value of a device with MAC=1, the device with MAC=2 would reply with a timeout, which did not occur in version 1.3.5
[image]

skarg, 2024-09-29:
Thank you for reporting the problem. To help me duplicate the problem:

- Can the MS/TP capture be uploaded here?  
- Which dlmstp.c module is the project using? 
- Were you able to determine the cause of the problem?

fangjj0709, 2024-09-30:
I'm sorry, I'm not sure what caused it yet, I only know that this issue does not exist in version 1.3.5, I am using the ports/linux/dlmstp_linux.c module.
[mstp_20240904153817.zip](https://github.com/user-attachments/files/17182704/mstp_20240904153817.zip)

fangjj0709, 2024-09-30:
> Thank you for reporting the problem. To help me duplicate the problem:
> 
> * Can the MS/TP capture be uploaded here?
> * Which dlmstp.c module is the project using?
> * Were you able to determine the cause of the problem?
I'm sorry, I'm not sure what caused it yet, I only know that this issue does not exist in version 1.3.5, I am using the ports/linux/dlmstp_linux.c module.
[mstp_20240904153817.zip](https://github.com/user-attachments/files/17182717/mstp_20240904153817.zip)

fangjj0709, 2024-09-30:
> 感谢您报告问题。为了帮助我重现此问题：
> 
> * 可以在这里上传 MS/TP 捕获吗？
> * 该项目使用哪个 dlmstp.c 模块？
> * 您能确定问题的原因吗？

I'm sorry, I'm not sure what caused it yet, I only know that this issue does not exist in version 1.3.5, I am using the ports/linux/dlmstp_linux.c module.
[mstp_20240904153817.zip](https://github.com/user-attachments/files/17182720/mstp_20240904153817.zip)

closed 2024-10-10 by skarg