bacnet-stack/bacnet-stack #786: Compile error using make router-mstp

deve87, 2024-09-27:
Tried to build the router-mstp and after a while gives me this error.

This from a OpenWRT platform

```/usr/bin/ld: main.o: in function 'datalink_send_pdu.constprop.0.isra.0':
main.c:(.text.datalink_send_pdu.constprop.0.isra.0+0x4c): undefined reference to 'dlmstp_send_pdu'
/usr/bin/ld: main.o: in function 'main':
main.c:(.text.startup.main+0xfc): undefined reference to 'dlmstp_set_max_info_frames'
/usr/bin/ld: main.c:(.text.startup.main+0x124): undefined reference to 'dlmstp_set_max_master'
/usr/bin/ld: main.c:(.text.startup.main+0x148): undefined reference to 'dlmstp_set_baud_rate'
/usr/bin/ld: main.c:(.text.startup.main+0x170): undefined reference to 'dlmstp_set_mac_address'
/usr/bin/ld: main.c:(.text.startup.main+0x17c): undefined reference to 'dlmstp_init'
/usr/bin/ld: main.c:(.text.startup.main+0x20c): undefined reference to 'dlmstp_get_my_address'
/usr/bin/ld: main.c:(.text.startup.main+0x2dc): undefined reference to 'dlmstp_receive'
/usr/bin/ld: main.c:(.text.startup.main+0x380): undefined reference to 'dlmstp_cleanup'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:28: router-mstp] Error 1
make[1]: Leaving directory '/root/bacnet-stack/apps/router-mstp'
make: *** [Makefile:449: router-mstp] Error 2
```

This from a debian running on a RPi

```main.c:(.text.datalink_send_pdu.constprop.0.isra.0+0x50): undefined reference to 'dlmstp_send_pdu'
main.c:(.text.datalink_send_pdu.constprop.0.isra.0+0x50): undefined reference to 'dlmstp_send_pdu'
/usr/bin/ld: main.o: in function 'main':
main.c:(.text.startup.main+0xf8): undefined reference to 'dlmstp_set_max_info_frames'
/usr/bin/ld: main.c:(.text.startup.main+0x120): undefined reference to 'dlmstp_set_max_master'
/usr/bin/ld: main.c:(.text.startup.main+0x144): undefined reference to 'dlmstp_set_baud_rate'
/usr/bin/ld: main.c:(.text.startup.main+0x16c): undefined reference to 'dlmstp_set_mac_address'
/usr/bin/ld: main.c:(.text.startup.main+0x178): undefined reference to 'dlmstp_init'
/usr/bin/ld: main.c:(.text.startup.main+0x1fc): undefined reference to 'dlmstp_get_my_address'
/usr/bin/ld: main.c:(.text.startup.main+0x2d4): undefined reference to 'dlmstp_receive'
/usr/bin/ld: main.c:(.text.startup.main+0x378): undefined reference to 'dlmstp_cleanup'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:28: router-mstp] Error 1
make[1]: Leaving directory '/home/pi/bacnet-stack/apps/router-mstp'
make: *** [Makefile:445: router-mstp] Error 2
```

stargieg, 2024-09-27:
Have you already looked at https://github.com/stargieg/bacnet-feed/tree/main/bacnet-stack?
I could add a packet router-mstp, but bacnet-stack-router does the same thing and have a luci gui.

deve87, 2024-09-29:
Thank you for reply.

I'm struggling with the openwrt packet build process. Can I as favor?  If it possible to add "bacnet-stack-router" for 23.05.3/arm_cortex-a7_neon-vfpv4/ ?

stargieg, 2024-09-29:
> Thank you for reply.
> 
> I'm struggling with the openwrt packet build process. Can I as favor? If it possible to add "bacnet-stack-router" for 23.05.3/arm_cortex-a7_neon-vfpv4/ ?

Yes. WiP ...

stargieg, 2024-09-29:
add this feed on your rpi
```
wget -O /etc/opkg/keys/152ccf91cd6bfbdd http://feeds.lunatiki.de/bacnet/releases/23.05.3/arm_cortex-a7_neon-vfpv4/152ccf91cd6bfbdd
echo 'src/gz bacnet http://feeds.lunatiki.de/bacnet/releases/23.05.3/arm_cortex-a7_neon-vfpv4' >> /etc/opkg/customfeeds.conf
opkg update
opkg install bacnet-stack-router
```
If you have luci installed
```
opkg install luci-app-bacrouter
```

deve87, 2024-09-29:
Great! Thank you.
Have now installed bacnet-stack-router with the luci app and will test it out.

deve87, 2024-09-29:
Have now run som test and it seems to be working. But not able to route a MS/TP device from Johnson Controls and only "Third-party devices" as J.C. calls them.

I know J.C devices runs more information than regular BACnet like cpu/mem.usage(etc) not as BACnet objects and can upload/download programs over this protocol.

stargieg, 2024-09-29:
The diff between the "router" and the "router-mstp" is the config and pthread.
Do you use the eXplorer from JCI (I still say JCI to Johnson Controls). If so i can try to reproduce this with more details.

deve87, 2024-09-29:
Yes, you are right. JCI is the right way.

I use the FEC/FAC and CG (black) series. I guest they are the same as facility explorer.

stargieg, 2024-09-30:
This package dump looks good to me. Can you capture the ip communication too?

I have build the router-mstp binary for rpi. No init script no UI.
```
opkg update
opkg install bacnet-stack-router-mstp
```
But I think it is the same.

deve87, 2024-09-30:
https://drive.google.com/file/d/1jd6KmG0PjgiZ93jgbku1e4cMl3az1A8x/view?usp=drive_link
The first part is when I use JCI autodiscover in a NAE network controller. Not able to discover the device there, but
when I manually added the device and tried to discover objects. No objects discovered, but device stays online and is recognized as  an JCI FEC controller.

192.168.10.1 is the mstp router and 192.168.10.102 is the JCI NAE network controller

stargieg, 2024-09-30:
There are some Confirmed-REQ confirmedPrivateTransfer with same Invoke ID.
Means it is repeated.
Timeout or Packet lost problem.

I can't test so well right now because I'm on the road.

deve87, 2024-10-03:
It works great for other products than JCI. Is it possible to also add the bacnet-router for 21.02.0 with the arm_cortex-a7_neon-vfpv4 ?
We use a lot of Teltonika RUTX08 routers and it runs a openwrt 21.02.0. System Target is Qualcomm Atheros IPQ4XXX.

stargieg, 2024-10-03:
Add repo public key
```
wget -O /etc/opkg/keys/152ccf91cd6bfbdd http://feeds.lunatiki.de/bacnet/releases/21.02.7/arm_cortex-a7_neon-vfpv4/152ccf91cd6bfbdd
```
Add repo
```
echo 'src/gz bacnet http://feeds.lunatiki.de/bacnet/releases/21.02.7/arm_cortex-a7_neon-vfpv4' >> /etc/opkg/customfeeds.conf
```

Not tested

deve87, 2024-10-03:
Thank you. It works

stargieg, 2024-10-03:
I have written building instructions for OpenWrt in the README.md.
https://github.com/stargieg/bacnet-feed/blob/main/README.md
Can we close the ticket with this?
Feel free to open a new ticket because of the proper JCI protocol. If I happen to have a JCI device in my hand again I will test it.

deve87, 2024-10-04:
Great, thank you again. Really appreciate you're help.

YABE where able to discover the JCI MSTP devices and if I set "Discover All As General BACnet Device = True" in the JCI BACnetIP integration on the JCI network controller. It also where able to discover these devices.

I can se the RX/TX light on my USB - RS485 adapter stops communication at some point when I connect a JCI MSTP to the MSTP bus. So the JCI MSTP controllers is doing something different than a "normal" BACnet MSTP device where the "blinking" is constant in between.

When this happens, the TX light stops blinking and only RX blinks. Wireshark then reports MAC address 5 (JCI device) "Poll For Master" from MSTP router MAC 10 address, and all the way up to 127. Before it starts communicate with the router again at address 10.

Maby set max master lower in the JCI MSTP device would "speed up" the communication. But other MSTP devices to not do this when it has established communication with the router.

closed 2024-10-05 by deve87