bacnet-stack/bacnet-stack #787: app bacepics misses optional parameters for the device object

jd-imi, 2024-09-27:
I have a device including those device properties:
* Location
* Description
* Serial Number
* Local Time
* Local Date
* UTC Offset
* DST

Those properties are not listed in the EPICS generated by bacepics. If I generate them with VTS, I can see those properties in the output.

I also notice that property-list is never shown whatever the object type (I don't know if this is linked)

skarg, 2024-09-27:
Thanks for reporting the problem.  It is likely caused by the device not being able to return ALL properties in a single ReadPropertyMultiple-Request (perfectly normal) and the EPICS tool fallback ReadProperty-Request not reading ALL individual properties (i.e. Required and Optional).

skarg, 2024-09-27:
I added a few more properties to the existing lists, but the properties you reported as missing were already in the optional list for the device object.

I tested the `bacepics` tool on the example `bacserv` after adding serial-number property to the server, and noticed that I needed to use "-o" optional parameter to enable the extra BACnet optional properties to be read from the device objects:
```
./bin/bacepics -v -o 260001 > epics.txt
```

skarg, 2024-09-27:
I'm researching what to do about the Property_List property that we added to the BACnet standard.  Property_List is mostly hidden as designed, but I see it referenced in EPICS in 135.1 and it will need to be added to `bacepics` as well.