bacnet-stack/bacnet-stack #793: app router is not working with makefile and BUILD=release

stargieg, 2024-10-01:
Compile with Makefile has OPTIMIZATION -Os (Size)
https://github.com/bacnet-stack/bacnet-stack/blob/c18768ba61402e73e11c61216691fa89d06cb825/apps/Makefile#L146
```
make clean;BUILD=release make router
bin/router -c ../bacrouter.cfg 
opt = c
dev_type = bip
dev_type = mstp
cmd file parse success
Initializing...
Interface: br0
IP Address: 10.230.6.103
IP Broadcast Address: 10.230.6.127
UDP Port: 0xBAC0 [47808]
RS485: Initializing /dev/ttyUSB0=success!
send to 10.230.6.127
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Aborted
```
Compile with OPTIMIZATION -O3 is in cmake and works
https://github.com/bacnet-stack/bacnet-stack/blob/c18768ba61402e73e11c61216691fa89d06cb825/CMakeLists.txt#L185
and Makfile with env OPTIMIZATION=-O3 works also.
```
make clean;BUILD=release OPTIMIZATION=-O3 make router
bin/router -c ../bacrouter.cfg 
opt = c
dev_type = bip
dev_type = mstp
cmd file parse success
Initializing...
Interface: br0
IP Address: 10.230.6.103
IP Broadcast Address: 10.230.6.127
UDP Port: 0xBAC0 [47808]
RS485: Initializing /dev/ttyUSB0=success!
send to 10.230.6.127
received from 10.230.6.103
```

Something is being optimized away with -Os.

skarg, 2024-10-02:
There seems to be *a lot* of discussion on other applications that have the exact same problem ([searched via Google](https://www.google.com/search?q=***+bit+out+of+range+0+-+FD_SETSIZE+on+fd_set+***%3A+terminated&sourceid=chrome&ie=UTF-8)). I didn't see a clear fix, other than indications either a) the file descriptor is invalid (i.e. -1), or b) there is memory corruption.

One of the suggestions is to run programs through strace, ptrace, truss or whatever your system offers, which should at least report the syscalls made which might be sufficiently revealing.

stargieg, 2024-10-02:
Can't figure out whats going on. But I can trace it back to the optimization of function dl_ip_thread().
2cea9727e292d87ff166a9da9d86a94be302ac31

PR #808 (2024-10-14): fix app router
with optimize Os the program exits with
bit out of range 0 - FD_SETSIZE on fd_set

Disable optimize for size by static set optimize 2

This Fix #793
commit b555e4a4 (2024-10-14): fix app router (#808)

with optimize Os the program exits with
bit out of range 0 - FD_SETSIZE on fd_set

Disable optimize for size by static set optimize 2

This Fix #793
closed 2024-10-14 by skarg