bacnet-stack/bacnet-stack #797: Not able to bind IP to bacrouter

deve87, 2024-10-06:
When I start bacrouter with a specific interface eth1. It then start 47808 with 0.0.0.0 and not as the actual IP of the eth1 network. 192.168.11.10
So I'm not able to start a second BACnetIP service for eth0 because the 47808 port is in use.

bacrouter-mstp do bind itself to the IP of the interface eth1 and then I can run a second BACnet service for eth0, but it don't work properly with JCI mstp device like bacrouter does.

Is there any way to specify an IP to the bacrouter?

With router-mstp
`udp        0      0 192.168.11.10:47808     0.0.0.0:*`

With router
`udp        0      0 0.0.0.0:47808           0.0.0.0:*`

stargieg, 2024-10-08:
You can't bin to an IP. because the router have to receive net broadcast on the interface.

I have the same Problem. I'm using ip veth and ip netns. I have a plan to implement this in openwrt init script.

closed 2024-10-14 by deve87

deve87, 2025-03-27:
Was able to make the router bind to the ethernet adapter IP only by changing

`sin.sin_addr.s_addr = htonl(INADDR_ANY);`

in ipmodule.c to

`sin.sin_addr.s_addr = ip_data->local_addr.s_addr;`

Still works fine and now able to have a BACnetIP server running on port 47808 reading from the MSTP network just by adding another IP to the interface.