bacnet-stack/bacnet-stack #798: Bacnet ms/tp problem with multiple devices.

feldsys, 2024-10-07:
I implemented the stack on a stm32g0,
If I use a single device on the rs485 bus, with just one bacnet router or Yabe everything is working fine I can  find the device in yabe or my bacnet viewer. Read/write any objects I implemented. 
If I add a second device of mine, (own Mac <127) and own Device ID I can't find anything. Sometimes one device is still here. Both have  the same code just id and Mac different. 
Can you give me some tips where can be the problem I just haven't any idea anymore. 

Thanks a lot in forward.

skarg, 2024-10-07:
The [mstpcap](https://steve.kargs.net/bacnet/bacnet-mstp-wireshark-live-capture/) utility, along with [Wireshark](https://wireshark.org/), can usually help diagnose problems on MS/TP segments.  Is it possible to get an MS/TP capture to narrow down the problem?

With new microcontroller 'ports' to MS/TP, I often encounter 'timing' issues with the 'end of transmit' that signals the RS485 DE/RE (RTS) to return to receive mode. An oscilloscope is useful to notice the end of the packet to be sure it is not truncated. I have not had good success using microcontroller peripherals that claim to have RS485 support as the feature uses a 'character' timeout out to control the RTS/DE pin rather than using an end of packet signal.

When a device is installed in field site segment, there can also be physical issues with wiring, polarity, and terminations, but that is probably not the problem if you are bringing up a new device.

feldsys, 2024-10-08:
Dear Steve,
 
Thank you for your quick reply. I have now recreated the behavior with the following parameters, router MAC is 0 and two devices with ID/MAC= 2 and ID/MAC=3, using Wireshark and learned the following:
1) My devices pass the token correctly most of the time but obviously there is an issue that the token suddenly stops being passed.
This heals itself after a while. For this I have added the Wireshark capture 003_2devices_1router_token-passing.pcapng.
2) When I read the objects with my router using a more complex query of (e.g. readPropertyMultiple), I do not get a valid response. See “002_2devices_1router_startup_and_error.pcapng” from line 556.
 
Thank you very much for your help, I hope you can give me a good tip on how to fix this error!
[waves_osci.zip](https://github.com/user-attachments/files/17292591/waves_osci.zip)
[wireshark.zip](https://github.com/user-attachments/files/17292592/wireshark.zip)

skarg, 2024-10-09:
The packets might not have enough Tpostdrive:

```
9.2.3 Timing
...
Transmitter enable: A node shall enable its EIA-485 driver before it generates 
the leading edge of the first start bit of a frame. The node shall drive the line
to the logical one state during the time between the enable and the leading
edge of the first start bit of a frame.

Transmitter disable: A node shall not disable its EIA-485 driver until the stop
bit of the final octet of a frame has been generated. The node shall disable
its EIA-485 driver within Tpostdrive after the beginning of the stop bit of the
final octet of a frame in order that it not interfere with any subsequent frame
transmitted by another node. This specification allows, but does not encourage,
the use of a "padding" octet after the final octet of a frame in order to facilitate
the use of common UART transmit interrupts for driver disable control. 
If a "padding" octet is used, its value shall be X'FF'. The "padding" octet is
not considered part of the frame, that is, it shall be included within Tpostdrive.

Receive to Transmit turn-around: A node shall not enable its EIA-485 driver
for at least Tturnaround after the node receives the final stop bit of any octet.
```

I also don't know your RS485 circuit, and if you are using isolated RS-485 there are likely opto-couplers involved.  There can be slow or fast opto-couplers.  I worked on an Atmel ATxmega 32MHz product (flow meter) in 2013 that used some slow opto-couplers for isolated RS-485 . I used the ports/bdk-atxx4-mstp/ code base to start, although the product was running on FreeRTOS.  I had to add a delay after turning on the RTS (enable transmitter) before sending any UART data.  I also had to delay the turn off the RTS after transmitting the packet, but not later than the MS/TP specification permits (Tpostdrive is15 bit times).  

Although there are elegant ways to do this (I did it with timer interrupts in the FreeRTOS project), just modify rs485.c and adjust the delay value until there is a clear RS-485 transmit enable signal (or if you know the amount of lag due to the opto coupler used in the design):
```
void rs485_rts_enable(
    bool enable)
{
    if (enable) {
        BIT_SET(PORTD, PD4);
    } else {
        BIT_CLEAR(PORTD, PD4);
    }
    _delay_us(10);
}
```
In the FreeRTOS project, I used _delay_ms(1) on the enable, and 3 character timeout on disable via timer interrupt.  On another isolated RS-485 project, I use Tpostdrive-5 (10 bit times) via a timer interrupt to trigger the transmitter disabled, but don't have any delay on enable since the optos are quite fast.

feldsys, 2024-10-10:
Over the past two days, I have thoroughly analyzed everything to the best of my knowledge and judgment. I’ve come to the conclusion that, in my opinion, RTS is not causing any issues.

During my analysis of mstpcap, I observed that when querying objects addressed to MAC address 2, MAC address 3 also responds, which should not happen. Upon reviewing the master state machine, particularly in the following cases:

```
case FRAME_TYPE_BACNET_DATA_EXPECTING_REPLY:
case FRAME_TYPE_BACNET_EXTENDED_DATA_EXPECTING_REPLY:
```
I noticed that the code only checks if the destination address is a broadcast. Shouldn't it also check if the request is addressed to This_Station?
```
 case FRAME_TYPE_BACNET_DATA_EXPECTING_REPLY:
                        case FRAME_TYPE_BACNET_EXTENDED_DATA_EXPECTING_REPLY:
                            if (mstp_port->DestinationAddress ==
                                MSTP_BROADCAST_ADDRESS) {
                                /* broadcast DER just remains IDLE */
                            } else if (mstp_port->DestinationAddress ==
                            		mstp_port->This_Station) {

                                /* indicate successful reception to the higher
                                 * layers  */
                                (void)MSTP_Put_Receive(mstp_port);
                                mstp_port->master_state =
                                    MSTP_MASTER_STATE_ANSWER_DATA_REQUEST;
                            }
                            break;
```
I added this condition, and now everything works perfectly.

Did I miss something?

skarg, 2024-10-10:
Thank you for the code analysis and suggested fix!  It is the correct fix, and I'll patch the code.

The original (and current standard) MS/TP Receive State machine checked TS (this station) prior to declaring 'ReceivedValidFrame'.  However, I modified the MS/TP Receive State machine to support ZeroConfig (choosing its own MAC address) and now every station message enters the Master Node State Machine, and the check for TS (this station) is missing.

skarg, 2024-11-27:
Fixed in [#803].