bacnet-stack/bacnet-stack #851: Parallel readprop commands scrambles the output

SindreLindahl, 2024-11-08:
Made a small program in node red to run multiple commands for readprop. Individually these work great, but when running a list of commands executed in parallel, it seems the output of each command gets mixed up and returns to a random command output?

For example: `/usr/local/bacnet-stack/build/readprop 111 2 2 description`

Adding delay or sequential execution solves the problem. Is this a known problem or limitation with bacnet stack?

skarg, 2024-11-08:
BACnet uses invoke-id along with device and service to match up replies (Ack) from a request. There is an optional Invoke-ID environment variable that you could set to uniquely identify each thread.

SindreLindahl, 2024-11-25:
Thanks! Sounds like a huge bug potential, imagine same behavior for any other transport protocol 😜