bacnet-stack/bacnet-stack #870: Reinitialize Device - requesting data from  handler_reinitialize_device in h_rd.c

DhanyasriR, 2024-12-10:
**REMOTE DEVICE MANAGEMENT SERVICES:**

**REINITIALIZE DEVICE:**

1.
\bacnet\basic\service\h_rd.c
We require the following status of the service requested in handler_reinitialize_device().

i. We need the STATE, BACNET_ERROR_CLASS and BACNET_ERROR_CODE (To understand if the request was Valid/Invalid).

ii. We also need to know the return value of datalink_send_pdu() - to understand if the ack has been sent or not sent. 

For Instance, after calling handler_reinitialize_device(), I need to check 
                              - what is the state and Error class & Error code to check if it is an invalid request
                              - check if the datalink_send_pdu() has sent/not sent the ack 
before calling the restart function in local.

**Could these data be exposed 
BACNET_REINITIALIZE_DEVICE_DATA rd_data ,   int len** 

from the void handler_reinitialize_device(
    uint8_t *service_request,
    uint16_t service_len,
    BACNET_ADDRESS *src,
    BACNET_CONFIRMED_SERVICE_DATA *service_data) function ?

2.
bool Device_Reinitialize(BACNET_REINITIALIZE_DEVICE_DATA *rd_data) in
\bacnet\basic\object\device.c

The above function supports for the following BACNET_REINITIALIZED_STATE - BACNET_REINIT_COLDSTART, BACNET_REINIT_WARMSTART and BACNET_REINIT_ACTIVATE_CHANGES.
But We only require Cold Start and Warm Start. 

BACNET_REINIT_ACTIVATE_CHANGES need not be supported. Is there any way for us to select only Warmstart and ColdStart States?

skarg, 2024-12-10:
To customize a 'bacnet/basic/service' simply start with the service (i.e. h_rd.c) and copy it locally to your project and compile it locally with the changes you need.  The same advice is for the `bacnet/basic/object` examples and device.c which almost certainly is customized for your project. We can also make changes to the public 'bacnet/basic/service' handlers to suit the wider audience if needed, but I'm not sure - from your description - what those changes would be.

For example, for #2, we could add some 'ENABLE' defines for the extra features within the bacnet/basic/object/device.c module, or tie the new features to a specific protocol revision, or just add a 'callback' so your product can do whatever it needs to do when it receives the ReinitailizeDevice message.

For 3: this library could use ERROR_CODE_SUCCESS for success but the data structures are not uniformly filled this way since ERROR_CODE_SUCCESS was added for a specific use in Error Class - SECURITY, SUCCESS - The operation was successful.  The Device_Reinitialize() is never called if there are errors decoding the request, and the Device_Reinitialize() is required to fill the error_code and error_class if there are errors during the application usage and return 'false'.

DhanyasriR, 2024-12-15:
Thank you for the inputs !

closed 2024-12-15 by DhanyasriR