bacnet-stack/bacnet-stack #873: No Data Received from COV Notifications (ESP32)

coskunyildirimcosel, 2024-12-11:
While using the BACnet stack on ESP32, I send SubscribeCOVRequest and receive SimpleAck responses, but I do not receive any COV notifications when values change. The SubscribeCOVRequest appears successful, but the expected notifications do not arrive.

### Esp32 Logs
`BIP: Broadcast Address: 192.168.1.255  
BIP: UDP Port: 0xBAC0 [47808]  
Analog Input 100: Value changed from 0.00 to 0.10  
Analog Input 100: Value changed from 0.10 to 0.20  
Analog Input 100: Value changed from 0.20 to 0.30  
Analog Input 100: Value changed from 0.30 to 0.40  
Analog Input 100: Value changed from 0.40 to 0.50  
SubscribeCOV: Sending Simple Ack!  
Analog Input 100: Value changed from 0.50 to 0.60  
Analog Input 100: Value changed from 0.60 to 0.70  
Analog Input 100: Value changed from 0.70 to 0.80  
Analog Input 100: Value changed from 0.80 to 0.90  
Analog Input 100: Value changed from 0.90 to 1.00  
Analog Input 100: Value changed from 1.00 to 1.10  
SubscribeCOV: Sending Simple Ack!  
Analog Input 100: Value changed from 1.10 to 1.20  
Analog Input 100: Value changed from 1.20 to 1.30  
Analog Input 100: Value changed from 1.30 to 1.40  
`

### Yabe Output

`SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
Sending SubscribeCOVRequest ...  
SimpleAck  
`
SubscribeCOVRequest sends successfully, and SimpleAck is received.
Despite value changes, no COV notifications are received.
The same behavior is observed with Yabe, where only SimpleAck responses appear, but no data is transmitted.

Bacnet-stack latest version 1.4.0

Can you help me with this?

skarg, 2024-12-11:
Assuming this is for a COV server device (i.e. the device replies to SubscribeCOV message and sends ConfirmedCOVNotifications and UnconfirmedCOVNotification messages) similar to apps/server/main.c example that is using COV server features.  Some code snippets from that example that are relevant to COV:
~~~
/* task timer for various BACnet timeouts */
static struct mstimer BACnet_Task_Timer;
/* task timer for TSM timeouts */
static struct mstimer BACnet_TSM_Timer;

static void Init_Service_Handlers(void)
{
...
    apdu_set_confirmed_handler(
        SERVICE_CONFIRMED_SUBSCRIBE_COV, handler_cov_subscribe);
...
    /* configure the cyclic timers */
    mstimer_set(&BACnet_Task_Timer, 1000UL);
    mstimer_set(&BACnet_TSM_Timer, 50UL);
}

int main(int argc, char *argv[])
{
    uint32_t elapsed_milliseconds = 0;
    uint32_t elapsed_seconds = 0;
...
    for (;;) {
...
        if (mstimer_expired(&BACnet_Task_Timer)) {
            mstimer_reset(&BACnet_Task_Timer);
            elapsed_milliseconds = mstimer_interval(&BACnet_Task_Timer);
            elapsed_seconds = elapsed_milliseconds / 1000;
            /* 1 second tasks */
            dcc_timer_seconds(elapsed_seconds);
...
            handler_cov_timer_seconds(elapsed_seconds);
...           
        }
        if (mstimer_expired(&BACnet_TSM_Timer)) {
            mstimer_reset(&BACnet_TSM_Timer);
            elapsed_milliseconds = mstimer_interval(&BACnet_TSM_Timer);
            tsm_timer_milliseconds(elapsed_milliseconds);
        }
        handler_cov_task();
    }
...
}
~~~
Note: DeviceCommunicationControl can disable initiation of COV notifications
Note: TSM is needed for Confirmed COV Notifications.

coskunyildirimcosel, 2024-12-12:
The problem was solved when I moved handler_cov_task() to another task. Thank you Steve for your quick feedback

closed 2024-12-12 by coskunyildirimcosel