bacnet-stack/bacnet-stack #902: Response is received when Confirmed service sent when DeviceCommunicationControl state is disable initialization but it should not work.

Singa17, 2025-02-04:
Issue 
When DCC_Enable_Disable is set to COMMUNICATION_DISABLE_INITIATION and device is responding to the confirmed services. But device shall not respond in this scenario.

Steps to reproduce:
1. Set password 
2. Send DeviceCommunicationControl Service with disable initialization indefinite time.
3. Send confirmed read property request.
4. Expect there shall be no response as per the service procedure of DeviceCommunicationControl Service.

But confirmed read property request is responds even though  DeviceCommunicationControl  state is in disable initialization.

Root Cause
there is no handling of confirmed services for disable initialization in incoming packet handler.  (apdu_confirmed_dcc_disabled).  But same is handled for unconfirmed service.

I could see there was a PR handling this but its got reverted. I am not sure why? Is it not a valid scenario?

Snapshot of DeviceCommunicationControl Service from Specification:
If the request is valid and the 'Enable/Disable' parameter is
DISABLE_INITIATION, the responding BACnet-user shall discontinue the initiation of messages except for I-Am requests
issued in accordance with the Who-Is service procedure and ConfirmedAuditNotification and UnconfirmedAuditNotification
requests.

skarg, 2025-02-04:
Disable Initiation means the ability to initiate a request (A side or client services - see Annex K for BIBBs that describe the -A and -B services).

The two types of application services messages are:

- `INITIATE` which corresponds to `-Request` confirmed message.  All unconfirmed messages are `INITIATE`.
- `EXECUTE` which corresponds  `-ACK` for confirmed messages.

Practical examples:

- a server device received a `ReadProperty-Request` and `DCC=DISABLE_INITIATION`, it will respond with `ReadProperty-ACK`
- a client-server device wants to send a `ReadProperty-Request` to another device and `DCC=DISABLE_INITIATION`, it cannot send the request.
- a server device received a `Who-Is-Request` and `DCC=DISABLE_INITIATION`, it will respond with `I-Am-Request`

Singa17, 2025-02-05:
Thanks for the quick reply.
 
We are server device (-B)
 
Now I am in alignment with your understanding.
 
As you rightly said, when the DCC is se to disable initiation, the server (B device) cannot initiate requests until client sends a request such as a read request, who-is. Also, it is understood from the ASHRAE 135-2020 Specification that I-AM devices should not be initiated from the Server (B device) when the DCC is set to Disable_Initiation.

In the current implementation of BACnet stack code, when the DCC is set to Disable_initiation on the server (B Device), then server (B device) is initiating the I-AM service anytime.
 
Which is right implementation for the Protocol revision 22 or greater. It would be good if you could advise on this behavior difference.

Code snippet from s_iam.c

void Send_I_Am(uint8_t *buffer)
{
    int pdu_len = 0;
    BACNET_ADDRESS dest;
    int bytes_sent = 0;
    BACNET_NPDU_DATA npdu_data;

#if 0
    /* note: there is discussion in the BACnet committee
       that we should allow a device to reply with I-Am
       so that dynamic binding always work.  If the DCC
       initiator loses the MAC address and routing info,
       they can never re-enable DCC because they can't
       find the device with WhoIs/I-Am */
    /* are we are forbidden to send? */
    if (!dcc_communication_enabled())
        return 0;
#endif

skarg, 2025-02-06:
`ANSI/ASHRAE Standard 135-2020` (protocol-revision 22) says:
```
16. REMOTE DEVICE MANAGEMENT SERVICES
DeviceCommunicationControl Service

16.1.2 Service Procedure

After verifying the validity of the request, including the 'Time Duration' and 'Password' parameters, the responding BACnet-
user shall respond with a 'Result(+)' service primitive. If the request is valid and the 'Enable/Disable' parameter is
DISABLE_INITIATION, the responding BACnet-user shall discontinue the initiation of messages except for I-Am requests
issued in accordance with the Who-Is service procedure and ConfirmedAuditNotification and UnconfirmedAuditNotification
requests. Communication shall be disabled in this manner until either the 'Time Duration' has expired or a valid
DeviceCommunicationControl (with 'Enable/Disable' = ENABLE) or, if supported, a valid ReinitializeDevice (with
'Reinitialized State of Device' = WARMSTART or COLDSTART) message is received.

If the responding BACnet-user does not have a clock and the 'Time Duration' parameter is not set to "indefinite," the APDU
shall be ignored and a 'Result(-)' service primitive shall be issued. If the 'Password' parameter is invalid or absent when a
password is required, the APDU shall be ignored and an Error-PDU with 'Error Class' = SECURITY and 'Error Code' =
PASSWORD_FAILURE shall be issued.

If the request is valid and the 'Enable/Disable' parameter is the deprecated value DISABLE, the request shall be ignored and
an Error-PDU with 'Error Class' = SERVICES and 'Error Code' = SERVICE_REQUEST_DENIED shall be issued.
```

`ANSI/ASHRAE Standard 135-2016` is similar but includes the deprecated `DISABLE` option specification (I think `DISABLE` was deprecated in protocol-revision 20).

Singa17, 2025-02-14:
Yes, Your right about DISABLE.
As per the spec protocol-revision 20 , when the disable initiation is set, device shall only reply when there is a who is request.
But i could see in s_iam.c file and Send_I_Am function , condition for the checking communication is enabled or not  is disabled. this makes device to send iam message even the dcc state is disable initiation.

**Our Scenario:**
Set the device to send "I am" at periodically (every 1 min)
Send DCC with disable Initiation.
Expecting there shall be no "I am" from the device.

Code snippet from s_iam.c file.
void Send_I_Am(uint8_t *buffer)
{
    int pdu_len = 0;
    BACNET_ADDRESS dest;
    int bytes_sent = 0;
    BACNET_NPDU_DATA npdu_data;

#if 0
/* note: there is discussion in the BACnet committee
that we should allow a device to reply with I-Am
so that dynamic binding always work. If the DCC
initiator loses the MAC address and routing info,
they can never re-enable DCC because they can't
find the device with WhoIs/I-Am /
/ are we are forbidden to send? */
if (!dcc_communication_enabled())
return 0;
#endif

skarg, 2025-02-18:
Thank you for providing your scenario for a use-case.

- I added a new function `Send_I_Am_Broadcast()` that is designed to be used used with the Who-Is handler and has no DCC restrictions.
- I added DCC Disable Initiation setting to the existing `Send_I_Am()` function for an application to send at startup or at other times.
- I hope your scenario is not the normal operation because that will generate a lot of broadcast traffic across the entire network

See pull request #918 for details.

closed 2025-02-26 by skarg