bacnet-stack/bacnet-stack #923: write_property_type_valid error value overwritten in two cases

piotrwiniarczyk-silvair, 2025-02-22:
In the `bool Network_Port_Write_Property(BACNET_WRITE_PROPERTY_DATA *wp_data)`
when handling PROP_FD_BBMD_ADDRESS and PROP_FD_SUBSCRIPTION_LIFETIME properties they types are checked with the write_property_type_valid function.
The write_property_type_valid function sets:
```
            wp_data->error_class = ERROR_CLASS_PROPERTY;
            wp_data->error_code = ERROR_CODE_INVALID_DATA_TYPE;
```
but in those mentioned above cases these are ignored and the following are used:
```
                wp_data->error_class = ERROR_CLASS_PROPERTY;
                wp_data->error_code = ERROR_CODE_VALUE_OUT_OF_RANGE;
```

I believe that this is a bug and the code similar to one for PROP_MAX_MASTER and PROP_MAX_INFO_FRAMES should be used.

https://github.com/bacnet-stack/bacnet-stack/blob/6486dfcacbf784d6811462acb819bae6a1255cbb/src/bacnet/basic/object/netport.c#L3896C2-L3896C35

piotrwiniarczyk-silvair, 2025-02-22:
PROP_MAX_MASTER and PROP_MAX_INFO_FRAMES are MSTP specific, but the code to read them (Network_Port_Read_Property) and write them (Network_Port_Write_Property) is not between defines as in Network_Port_Read_Range.
```
#if defined(BACDL_MSTP)
        case PROP_MAX_MASTER:
        case PROP_MAX_INFO_FRAMES:
#endif 
```

skarg, 2025-02-24:
Thank you for reporting the problems. 

I added ifdef around Max-Master and Max-Info-Frames properties as suggested to e consistent. 
I fixed WriteProperty error handling for PROP_FD_BBMD_ADDRESS and PROP_FD_SUBSCRIPTION_LIFETIME properties as suggested.  See #925 pull reqeust.

closed 2025-03-03 by skarg