bacnet-stack/bacnet-stack #924: Write_Property error handling for some objects can be improved

piotrwiniarczyk-silvair, 2025-02-22:
For some of the objects the bacapp_decode_known_property function is used to check if an property is supported.
If it is not supported, the ERROR_CLASS_PROPERTY and ERROR_CODE_VALUE_OUT_OF_RANGE error is generated.

_VALUE_OUT_OF_RANGE - An attempt has been made to write to a property with a value that is outside the range
of values defined for the property._ 

While UNKNOWN_PROPERTY seems to be more appropriate:

_UNKNOWN_PROPERTY - A Property_Identifier has been specified in a service parameter that is unknown or
unsupported in the addressed BACnet device for objects of the referenced object type._

This observation is for color_object, color_temperature, lc, lo , netport.

For other objects, the bacapp_decode_application_data function is used and also ERROR_CODE_VALUE_OUT_OF_RANGE is generated. I think same logic can be applied and UNKNOWN_PROPERTY seems to fit better for those cases.

skarg, 2025-02-24:
The 'known' properties for WriteProperty in an object are those properties that are supported by ReadProperty/RPM/ReadRange. `bacapp_decode_known_property()` is trying to decode the context and application tagged property values where `bacapp_decode_application_data()` just tries the application tagged values, whether they are supported by the object or not.  The known/unknown is handled in the `switch()` statement in the WriteProperty handler.

Is there a specific property that is generating Out-of-Range when it should be generating Unknown-Property? Is there a property where the logic is determined by `bacapp_decode_known_property()` instead of switch statement in write property handling?

piotrwiniarczyk-silvair, 2025-02-25:
This report is purely based on reading the error handling in the specification. The error handling section of the specification is giving description of general errors while the service section is giving description of errors in the context of a service. The UNKNOWN_PROPERTY seems to fit better here. I would expect that error checking is using a table from 15.9.1.3.1 Error Type from top to down. I understand that top to down is not a requirement, but my feeling is that it was intended that way.

UNKNOWN_OBJECT - unknown object
UNKNOWN_PROPERTY - known object, but unknown property
PROPERTY_IS_NOT_AN_ARRAY + other array - known object, known property, but an array error
WRITE_ACCESS_DENIED - known object, known property, index/array ok, but property is R
INVALID_DATATYPE - known object, known property, index/array ok, property is W, but a packet contains wrong datatype for an identified property
DUPLICATE_NAME, DUPLICATE_OBJECT_ID - known object, known property, index/array ok, property is W, a packet contains good datatype, but a value cannot be set due to "duplication" requirement
VALUE_OUT_OF_RANGE - known object, known property, index/array ok, property is W, a packet contains good datatype, but a value cannot be set due to out of range requirement

This thinking is not reflected in Network_Port_Write_Property code. While some of the errors are out of scope, the second error code UNKNOWN_PROPERTY is not generated on second step of checking. 

I do not know if test specification is also following such reasoning , but I will try to look for tests in this area.

`bool Network_Port_Write_Property(BACNET_WRITE_PROPERTY_DATA *wp_data)
{
    bool status = false; /* return value */
    int len = 0;
    BACNET_APPLICATION_DATA_VALUE value = { 0 };

    if (!Network_Port_Valid_Instance(wp_data->object_instance)) {
        wp_data->error_class = ERROR_CLASS_OBJECT;
        wp_data->error_code = ERROR_CODE_UNKNOWN_OBJECT;
        return false;
    }
    /* decode the some of the request */
#if defined(BACAPP_COMPLEX_TYPES)
    len = bacapp_decode_known_property(
        wp_data->application_data, wp_data->application_data_len, &value,
        wp_data->object_type, wp_data->object_property);
#else
    len = bacapp_decode_application_data(
        wp_data->application_data, wp_data->application_data_len, &value);
#endif
    if (len < 0) {
        /* error while decoding - a value larger than we can handle */
        wp_data->error_class = ERROR_CLASS_PROPERTY;
        wp_data->error_code = ERROR_CODE_VALUE_OUT_OF_RANGE;
        return false;
    }`

piotrwiniarczyk-silvair, 2025-02-28:
I am really confused here. Today I try an example implementation of  "9.22.1.X3 Writing NULL to Non-commandable Properties". It does not pass for name property of the object. On the other hand I not know if 9.22.1.X3 is needed at all.

I am still trying to understand what is required and what is not.

Another example is  /* Workaround BTL Specified Test 9.23.2.X5 */ in h_wpm.c
It is preventing to pass 9.22.2.3 with the test implementation I am using.

Going forward without automatic 'negative' tests (9.22.2, 9.23.2) in this area will be IMO a problem. One fix can easily break other test cases. 

So I cannot give a reasonable answer to your question. Something that looked simple at a first glance seems to be more complicated.

skarg, 2025-02-28:
For  "9.22.1.X3 Writing NULL to Non-commandable Properties" there is a different pull-request #919