bacnet-stack/bacnet-stack #948: bip interface not reconfigurable on Windows port

huhtams, 2025-03-21:
I am setting up a bacnet client for Windows using bip. I would want to switch the bound network interface on the client during program runtime. However, it seems to be not possible to reset the interface using e.g. the following code:
```
char addr1[] = "192.168.1.1";
bip_init(addr1);
bip_cleanup();
char addr2[] = "192.168.100.100";
bip_init(addr2); // Still uses the interface with address 192.168.1.1
```
For some reason there appears to be this guard which allows initializing the address only once to a non-zero value:
https://github.com/bacnet-stack/bacnet-stack/blob/8028d7b79abf831561558893d4760786bff75d7d/ports/win32/bip-init.c#L748-L753
I cannot really bypass this, since my only access to setting this static variable seems to be through this function. Is this by design, or could the if statement be dropped?

skarg, 2025-03-21:
Thank you for reporting the problem!

The check for non-zero `BIP_Address.s_addr` was designed to allow a statically configured address, but all the API for setting the address have been removed for those OS based examples:
```
/**
 * Set the BACnet/IP address
 *
 * @param addr - network IPv4 address
 */
bool bip_set_addr(const BACNET_IP_ADDRESS *addr)
{
    (void)addr;
    /* not something we do here within this application */
    return false;
}
```

For the OS based apps, it might make better sense to have `bip_cleanup()` also set `BIP_Address.s_addr = 0;`. Would that fix the problem?

huhtams, 2025-03-24:
I see, that would work. And maybe also `BIP_Broadcast_Addr.s_addr` needs to be reset at the same time?

closed 2025-03-25 by skarg