bacnet-stack/bacnet-stack #964: `dlenv_init` cannot choose among `BACDL_BIP`, `BACDL_MSTP`, and `BACDL_BSC`

ryanjmulder, 2025-04-10:
I'm working toward user-selectable datalink in my device and looking for direction on this issue.

Should I:
  1. Fix `dlenv_init`?
  2. Compile multiple copies of my bacnet server with different datalink selections?

`dlenv_init` seems to support choosing from among compiled-in datalinks using `getenv("BACNET_DATALINK")`

However, the MSTP configuration is guarded by `#elif defined(BACDL_MSTP)`, which seems like an easy fix - change to `#if` instead of `#elif`.

Similary, the BSC config is guarded by `#elif defined(BACDL_BSC)`, but the same fix doesn't work, because the call to `bacnet_secure_connect_network_port_init` calls `Network_Port_Type_Set` which could conflict with the `dlenv_init`'s final call to `dlenv_network_port_init`

Finally,  `dlenv_network_port_init` has various preprocessor-selected implementations (as opposed to selected via `getenv("BACNET_DATALINK")`)

skarg, 2025-04-13:
I made some changes to dlenv.c and the build files to better support multiple datalinks `BACDL_MULTIPLE` compile flag using `BACNET_DATALINK` environment variable in pull-request [#966].  Can you test the changes to see if they do what you were expecting?

ryanjmulder, 2025-04-14:
Thanks, Steve!
That looks like exactly what I was after. I'll test it as soon as I can.

ryanjmulder, 2025-04-15:
I tested with IP and MS/TP and it's working as expected. Thanks very much!