bacnet-stack/bacnet-stack #970: How to get the examples for MSTP

Hooli77, 2025-04-17:
Sorry if this is a trivial question, but how can I compile the stack so that the examples in the apps folder are for MSTP instead of IP?
I'm working on Windows 10.

Thanks in advance!

skarg, 2025-04-17:
The library uses a define `BACDL_MSTP` to enable MS/TP builds, and it can be done with other datalinks by choosing `BACDL_MULTIPLE` and picking the datalink you want the app/ to use.  See recent issue [#964] for information about choosing a datalink at runtime.

I usually use the Makefile (GNU Make) build system to build the apps/ for Windows.  I usually build from Linux using cross compiling (see the recipe in the github workflow), but GNU Make can be invoked from Windows under MSYS and MinGW.  There are details about MSYS and MinGW in the [ports/win32/README.md](https://github.com/bacnet-stack/bacnet-stack/blob/master/ports/win32/README.md) file.

The root Makefile has some shortcut recipes:
```
$ make mstpwin32
```
which gets invoked as:
```
$(MAKE) BACDL=mstp BACNET_PORT=win32 -s -C apps all
```

There are also CMakeLists.txt files that you can leverage to build the apps or unit tests or specific ports under Visual Studio Code, for example. You would need to choose the `BACDL_MSTP` option for the build.

Hooli77, 2025-04-17:
Thank you!

I have another question, concerning the object creation, more specifically, the device one. When I use Device_Init(Null), I see a lot of objects on Yabe, how can I get rid of the ones I dont need ?

skarg, 2025-04-17:
The default device table for the example server uses both fixed numbers of objects for some objects, and objects that can be created for other objects. You can define a custom 'device table' with only objects that include CreateObject features and pass the table into Device_Init.  See [apps/server-mini](https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server-mini/main.c) for an example.

Alternately, you can use the [apps/server-basic](https://github.com/bacnet-stack/bacnet-stack/blob/master/apps/server-basic/main.c) approach which uses [src/bacnet/basic/server](https://github.com/bacnet-stack/bacnet-stack/blob/master/src/bacnet/basic/server/bacnet_device.c) a general server and device with creatable objects.