bacnet-stack/bacnet-stack #972: How to create an MS/TP device in Linux

leexx111, 2025-04-21:
Excuse me. I'm a novice in bacnet. I'd like to ask how to create an MS/TP device in a Linux environment where its objects are mapped in a table of an SQLite database. Which files in the protocol stack are needed? thank you！

leexx111, 2025-04-21:
Previously, I attempted to modify the definition of the MS/TP communication port in rs485.h to "/dev/ttys2" (because this serial port is currently externally connected to an RS485 circuit), and then I make mstp , run the example apps/server-mini , but I cant scan this device on this serial port. After that, I didn't know what to do so

skarg, 2025-04-24:
The common port modules for OSes (Windows, Linux, BSD/macOS) use environment variables to configure settings for the various drivers. For MS/TP, the defaults are:
```
$ export BACNET_IFACE=/dev/ttyUSB0
$ export BACNET_MAX_INFO_FRAMES=1
$ export BACNET_MAX_MASTER=127
$ export BACNET_MSTP_BAUD=38400
$ export BACNET_MSTP_MAC=1
```
The defaults can be changed by using the set/export on OS that support environment variables. 
There is an example [router-mstp script](https://github.com/bacnet-stack/bacnet-stack/blob/master/bin/router-mstp.sh) that can be used prior to launching apps/router-mstp/ and some of the setting in the script could be used with other apps that only use the MS/TP datalink.