bacnet-stack/bacnet-stack #979: bug in Routed_Device_GetNext()

jshenguru, 2025-04-29:
The Routed_Device_GetNext() function in the gw_device.c is using MAX_NUM_DEVICES to check if the index is out of range. When the actual routed devices are less than the MAX_NUM_DEVICES, this would end up with "ghost" device with ID of 0 in the "I-am" response. 

To fix that, the Num_Managed_Devices should be used to check if the index is out of range.

skarg, 2025-04-29:
Thank you for reporting the problem.

I make a fix in pull-request [#981].  Can you verify that this change fixes the problem you observed?

jshenguru, 2025-04-29:
The PR only replaced the first occurrence of the MAX_NUM_DEVICES while there are other two in the function. I would think that it's better to replaced all three occurrences of the MAX_NUM_DEVICES in that function with Num_Managed_Devices (which is what I did in my test and it worked).

closed 2025-05-06 by skarg