bacnet-stack/bacnet-stack #998: Test BIT01300 fails, BDT empty

GauiStori, 2025-05-16:
I have been testing my device with the BACtest application and it fails on the BIT01300 test.

The BDT is empty. The bug can be reproduced in bacserv v1.4.1

When bacserv is started, there is a correct entry in the BDT but it seems to disappear in some
writing test that I haven't isolated yet.

I tried to disable the writing in the function 
bvlc_decode_write_broadcast_distribution_table
and that seems to solve the problem but that is not a solution.

skarg, 2025-05-16:
Thank you for reporting the problem.

There is very minimal error handling prescribed or implemented for the BVLC functions so `bvlc_decode_write_broadcast_distribution_table()` just sets an BDT entry to 'invalid' when the Write-Broadcast-Distribution-Table is sent and an entry has invalid data that is detected by the decoder. I suppose it could be smarter and verify that ALL entries in the Write are valid before writing them.

However, the test is probably checking to see if the device is honoring the change in operation in Protocol_Revision 17.
```
J.2.2.1 Write-Broadcast-Distribution-Table: Format
...
Prior to the introduction of the Network Port object in Protocol_Revision 17, 
this message was the interoperable means of updating BDTs. 
That function is now performed by writes to the Network Port object.

J.4.4.2 Use of the BVLL Write-Broadcast-Distribution-Table Message
Upon receipt of a BVLL Write-Broadcast-Distribution-Table message, 
B/IP devices shall always return a BVLC-Result message to the 
originating device with a result code of X'0010' indicating that 
the Write-Broadcast-Distribution BVLL message is not supported.
```

We'll need to add the check for protocol-revision and return the BVLC-Result of `BVLC_RESULT_WRITE_BROADCAST_DISTRIBUTION_TABLE_NAK` when greater than 17.

GauiStori, 2025-05-18:
Thanks Steve

I have still not found what test deletes the BDT but i can create problem with the bacwbdt utility. Making a list of equal IP-addresses/ports not containing the IUT's own address is accepted. 
Unfortunately I am not fully sure how the error detection shall be implemented. I found out that for my own device I should have compiled it as BBMD=client and that solves the problem for me. I may take a look at it later.

Regards
Gudjon

closed 2025-06-10 by skarg