bacnet-stack/bacnet-stack #1033: Polling data in a loop

tav74, 2025-06-30:
When receiving data from a device in a loop, an error periodically occurs in the bip_init_windows function on the line
Result = WSAStartup((1 << 8) | 1, &wd);

And also in the getIpMaskForIpAddress function on the line
DWORD dwStatus = GetAdaptersInfo(AdapterInfo, &dwBufLen);

How to correctly free memory in the BACnet Stack when cyclically receiving values ​​and re-receive values?

skarg, 2025-07-17:
What is the Windows error that is occurring?
See https://learn.microsoft.com/en-us/windows/win32/api/winsock2/nf-winsock2-wsastartup

tav74, 2025-07-21:
Thanks for the answer. The error was like accessing uninitialized data. But maybe it's from insufficient reading of documentation for your one of the best libraries. I made a division into data initialization and data retrieval

void QtBACnet::DoInit(QString myinterface, QString mydeviceAddress, QString myobjectType, QString myobjectNumber, QStringList myоbjectPropertys)

///////////////////////////////////////////////////////////////////////////////
Device_Init(NULL);

ba = myinterface.toLocal8Bit();
const char* myiface = ba.data();
bip_set_interface(myiface);

Init_Service_Handlers();
dlenv_init(); //improve so it doesn't crash
atexit(datalink_cleanup);

//////////////////////////////////////////////////////

and cyclic reading

void QtBACnet::DoRead()

/////////////////////////////

time_t last_seconds = 0;
time_t timeout_seconds = 0;
time_t current_seconds = 0;
time_t elapsed_seconds = 0;

last_seconds = time(NULL);
timeout_seconds = (apdu_timeout() / 1000) * apdu_retries();

QStringList object_prop;
QJsonArray my_arr;

found = address_bind_request(Target_Device_Object_Instance, &max_apdu, &Target_Address);

if (!found) { 
Send_WhoIs(Target_Device_Object_Instance, Target_Device_Object_Instance);
}

for (;;)
{ 
current_seconds = time(NULL); 

if (current_seconds != last_seconds) 
{ 
tsm_timer_milliseconds((uint16_t)((current_seconds - last_seconds) * 1000)); 
datalink_maintenance_timer(current_seconds - last_seconds); 

} 
if (Error_Detected) { 
break; 
} 

if (!found) { 
found = address_bind_request(Target_Device_Object_Instance, &max_apdu, &Target_Address); 
} 

if (found) 
{ 
if (Request_Invoke_ID == 0) 
//Request_Invoke_ID = Send_Read_Property_Request(Target_Device_Object_Instance, Target_Object_Type, Target_Object_Instance, Target_Object_Property, Target_Object_Index); 
Request_Invoke_ID = Send_Read_Property_Multiple_Request(&buffer[0], sizeof(buffer), Target_Device_Object_Instance, Read_Access_Data);
//////////////////////////////////////////////////////////////////////

the only problem is how to get other data in the cycle, from other branches of the object, without:

delete myQtBACnet;
myQtBACnet = NULL;
myQtBACnet = new QtBACnet;

but probably this is a feature of the protocol itself...

thanks again for your answer.