bacnet-stack/bacnet-stack #1044: How to set the simplest state of BACnet stack code

Junedoubling6, 2025-07-13:
At present, **I need to evaluate chip resources**, so I need to add the required functional objects from **the simplified state** of the Bacnet stack. Please tell me what parameters or code need to be set or blocked, or what precompilation conditions need to be canceled for **the simplified state** of the Bacnet stack. Or tell me which routine is closest to the simplest state of Bacnet stack. And please forgive any unclear expressions, as my native language is not English

skarg, 2025-07-13:
Which datalink (BACnet MS/TP, IPv4, IPv6, Ethernet, or Secure Connect) do you want to use with your chip?
What size APDU (50-1497) do you want to use?
Which objects and services do you want to use?
Which build system do you want to use?
https://github.com/bacnet-stack/bacnet-stack-zephyr 
There many examples of bare metal BACnet servers in `ports/` folder, and many examples of OS based servers in `apps/` folder, and examples of RTOS based profiles in [BACnet-Stack Zephyr repo](https://github.com/bacnet-stack/bacnet-stack-zephyr).

Junedoubling6, 2025-07-14:
The data link type has been selected as MS/TP.The APDU size is 480.

Junedoubling6, 2025-07-14:
I will describe the current code status and issues below: 
The data link type has been selected as MS/TP.
The APDU size is 480.
By referring to the apps/server-basic routine, I created 26 AI type objects and 19 AO type objects. Object service attributes include type, instance, units, value, * name, * description.
The compilation environment is keil mdk. The code is developed based on bare metal. The hardware platform chip model is GD32.

Also, I have some questions:
I have defined the following precompilation conditions: 
BACDL_MSTP=1, BACAPP_MINIMAL, MAX_APDU=480, BIG_ENDIAN=0, MAX_TSM_TRANSACTIONS=0, MAX_CHARACTER_STRING_BYTES=64, MAX_OCTET_STRING_BYTES=64, BACNET_STACK_DEPRECATED_DISABLE, MAX_ADDRESS_CACHE=64. Is there a need to add, delete or modify values?

I don't need network functionality, but I see that each object calls the API: Network_Port_Index_To_Instance(0) when it is created. May I ask if it is possible to block it?
And is there anything else that needs to be deleted or modified (such as replacing the file bacnet/basic/object/netport. c)

Can non essential other objects and related header files be blocked? (I see that the comparison of RAM usage before and after blocking is very obvious)

Junedoubling6, 2025-07-14:
Please inform me of any actions that are beneficial for optimizing resource utilization in my current state. Especially the occupation of SRAM. Look forward to your reply!